LibreCAD/LibreCAD #950: Printing a drawing is very faint, hardly visible.

N0madism, 2018-02-18:
Drawing looks good on screen.

When printing, print preview looks fine too, but when printed to my Brother HL-L2360D, it comes out very faint on paper. 

I printed same file through CutePDF to PDF format file, then printed from Adobe reader. 
Printout is perfect. just like on screen in LibreCAD. 

Using:
LibreCAD Version: 2.1.3
Compiler: GNU GCC 4.9.1
Compiled on: Sep 23 2016
Qt Version: 5.4.1
Boost Version: 1.60.0
System: Windows XP

Any ideas?
I haven't got any other printer to try with so can't exclude printer drivers, etc.

Swulf, 2018-03-18:
Have you looked at the Properties of the Layer on which the drawing is made? (Layer '0' is the default unless you created more)? There is a Pen Size or something that controls the thickness of the line used to draw elements on the layer. You can change it to something thicker/heavier and that should help your issue.

flywire, 2019-11-30:
As shown in [Layer Settings Default Pen Width](https://github.com/xanderdin/LibreCAD_dxf2pdf_standalone/issues/2#issuecomment-526479333).

https://user-images.githubusercontent.com/6764484/63998664-e641f980-cb01-11e9-9eea-2de7472c047b.png