LibreCAD/LibreCAD #955: lib/engine/rs_dimangular.cpp:144:69: error: ‘QStringLiteral’ was not declared in this scope

717g301github, 2018-03-13:
~~~
lib/engine/rs_dimangular.cpp: In member function ‘virtual QString RS_DimAngular::getMeasuredLabel()’:
lib/engine/rs_dimangular.cpp:144:69: error: ‘QStringLiteral’ was not declared in this scope
     int dimaunit {getGraphicVariableInt( QStringLiteral( "$DIMAUNIT"), 0)};
                                                                     ^
make[2]: *** [Makefile:39798: ../../generated/librecad/obj/rs_dimangular.o] Error 1
make[2]: Leaving directory '/home/data1/protected/Programming/git/LibreCAD/librecad/src'
make[1]: *** [Makefile:39: sub-src-make_default] Error 2
make[1]: Leaving directory '/home/data1/protected/Programming/git/LibreCAD/librecad'
make: *** [Makefile:78: sub-librecad-make_default-ordered] Error 2
~~~

Can somebody help me solve this?

lordofbikes, 2018-03-13:
Maybe, if you add some useful details to your request.
OS? Qt version? Compiler version? Source code version?

717g301github, 2018-03-14:
Compiler: g++ (GCC) 5.3.0
Qt, I have 5.5.1 on system, but qt 4.8.7 is used by "make". If you can help me adjust that one?

source code is from git

~~~~
/opt/qt4/bin/uic ui/forms/qg_dlgarc.ui -o ../../generated/librecad/ui/ui_qg_dlgarc.h
/opt/qt4/bin/uic ui/forms/qg_dlgcircle.ui -o ../../generated/librecad/ui/ui_qg_dlgcircle.h
/opt/qt4/bin/uic ui/forms/qg_dlgdimension.ui -o ../../generated/librecad/ui/ui_qg_dlgdimension.h
/opt/qt4/bin/uic ui/forms/qg_dlgdimlinear.ui -o ../../generated/librecad/ui/ui_qg_dlgdimlinear.h
/opt/qt4/bin/uic ui/forms/qg_dlgline.ui -o ../../generated/librecad/ui/ui_qg_dlgline.h
/opt/qt4/bin/uic ui/forms/qg_dlgellipse.ui -o ../../generated/librecad/ui/ui_qg_dlgellipse.h
g++ -c -pipe -g -std=c++11 -fext-numeric-literals -O2 -Wall -W -D_REENTRANT -DDWGSUPPORT -DQC_APPDIR="librecad" -DLC_VERSION="2.2.0-rc1" -DQT_NO_DEBUG -DQT_SVG_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/opt/qt-4.8.7/mkspecs/linux-g++ -I. -I/opt/qt-4.8.7/include/QtCore -I/opt/qt-4.8.7/include/QtGui -I/opt/qt-4.8.7/include/QtSvg -I/opt/qt-4.8.7/include -I/usr/include -I../../libraries/muparser/include -I../../libraries/libdxfrw/src -I../../libraries/jwwlib/src -Icmd -Ilib/actions -Ilib/creation -Ilib/debug -Ilib/engine -Ilib/fileio -Ilib/filters -Ilib/generators -Ilib/gui -Ilib/information -Ilib/math -Ilib/modification -Ilib/printing -Iactions -Imain -Itest -Iplugins -Iui -Iui/forms -Iui/generic -I../res -I../../generated/librecad/moc -I../../generated/librecad/ui -o ../../generated/librecad/obj/rs_dimangular.o lib/engine/rs_dimangular.cpp
lib/engine/rs_dimangular.cpp: In member function ‘virtual QString RS_DimAngular::getMeasuredLabel()’:
lib/engine/rs_dimangular.cpp:144:69: error: ‘QStringLiteral’ was not declared in this scope
     int dimaunit {getGraphicVariableInt( QStringLiteral( "$DIMAUNIT"), 0)};
                                                                     ^
make[2]: *** [Makefile:39798: ../../generated/librecad/obj/rs_dimangular.o] Error 1
make[2]: Leaving directory '/home/data1/protected/Programming/git/LibreCAD/librecad/src'
make[1]: *** [Makefile:39: sub-src-make_default] Error 2
make[1]: Leaving directory '/home/data1/protected/Programming/git/LibreCAD/librecad'
make: *** [Makefile:78: sub-librecad-make_default-ordered] Error 2
~~~~

lordofbikes, 2018-03-14:
> but qt 4.8.7 is used by "make"

Since LibreCAD 2.2 Qt5 is mandatory.

So you have to remove Qt4 or setup your environment correct to use both versions of Qt side by side.
This is a common issue and you will find lots of solutions out there, e.g. on https://stackexchange.com.

closed 2018-03-14 by lordofbikes