LibreCAD/LibreCAD #957: Is there any team or individual that try to improve LibreCAD's support for Chinese fonts?

qium06, 2018-03-17:
I am a Chinese engineer who often uses CAD, and I am also a part-time C++ programmer. I love LibreCAD and its spirit of freedom. But LibreCAD's support for Chinese fonts is not good which limits its practical use. So I've been trying to do something about it. Recently I've been studying LibreCAD's source code for several months. But the progress is slow. Is there any team or individual doing similar things? Please let me know. I'd like to join you! Thank you.

lordofbikes, 2018-03-21:
dxli worked on Chinese font support couple years ago. Sadly he has to quit contribution for a while now. So nobody is actively working on this currently.
As I understand his blog post http://blog.librecad.org/2011/11/complete-cjk-font-support-in-librecad/ the file size is reason that the font isn't supplied with the installer packages.

In the forum are a few posts too: http://forum.librecad.org/template/NamlServlet.jtp?macro=search_page&node=4499512&query=chinese+font

There were also ideas to implement TTF support, but this is not an easy task and nobody started with this.

You're more than welcome to support us with this. Maybe you can check the archive from the blog post and if it's valid, we can find a better download location.

dxli, 2026-10-07:
WenQuanYi and Noto fonts are now included in the 2.2.2-alpha/master branch

closed 2026-10-07 by dxli