LibreCAD/LibreCAD #960: Feature request - Ability to set the style of a POINT

Swulf, 2018-03-18:
It seems that a POINT added to a drawing (using the POINT command) can only have one style. When adding points that overlay lines, they are not visible. It would be better if the point style could be defined (a cross, a circle, ...)

lordofbikes, 2018-03-21:
There is no parameter for shape of point entity in DXF reference. So its hard to implement a solution compatible with other CAD software.

Why won't you use blocks for this? You can create your own shapes, with block reference point in the center and place it on the point coordinates. This would also be compatible with other CAD software.

paolo-caroni, 2020-05-13:
> There is no parameter for shape of point entity in DXF reference. So its hard to implement a solution compatible with other CAD software.
>
This isn't totally true.
I have helped to document properly DXF file for a library to read and write it.
https://github.com/bert/libdxf
If you read the doc you can read much useful info for DXF format.
In specific case read this https://github.com/bert/libdxf/blob/master/doc/header/PDMODE.html

Cheers,
Paolo

pollywinters, 2022-03-13:
I now have an implementation for this that is fully working. Very useful it has been as well in just the first couple of drawings I've done since finishing debugging it!
Unfortunately real life distractions are intervening at the moment and it may be some time before I can figure out writing a pull request etc. Anybody that is interested though, all the code is checked into my fork of the LibreCAD repository and you are totally welcome to look through and cherrypick. 
When I do get time I will write up whatever to make the changes generally accessible to all...

pollywinters, 2022-03-28:
Pull request created
#1532

sand1024, 2026-09-26:
too old, style of points is supported in drawing settings

closed 2026-09-26 by sand1024