LibreCAD/LibreCAD #961: Bug - CIRCLE (Center, Radius) doesn't work correctly

Swulf, 2018-03-18:
If you select the CIRCLE (Center, Radius) command and enter the center on the command line (i.e. 0,0) it never follows up by asking you the radius. It just continues to ask for the center.

Swulf, 2018-03-18:
OK, now I see that in fact it asks for the radius in a box at the top of the window rather than in the Command Line area. This seems like very weak and certainly counter-intuitive user interface design. At least, it should accept a radius from the command line area in the first instance, especially because if you entered the CIRCLE command in that window you would likely expect to be asked for follow-up information in the same location, not elsewhere on the screen.

Further, the behaviour is different between CIRCLE, CENTER, POINT and CIRCLE, CENTER, RADIUS! The first asks for all the necessary info from the command line area, the second asks you for the information from two different areas of the screen. Makes no sense.

lordofbikes, 2018-03-21:
Both tools work as designed. Use Circle-Center-Radius to draw many circles with same radius and Circle-Center-Point to draw many circles with different radius or in conjunction with snapping on other entities. That makes absolutely sense for me.
The tool options bar is used to request tool parameters by many other tools too. It's not a great moment in GUI design, but once recognized, users are aware of it.

closed 2018-03-21 by lordofbikes