LibreCAD/LibreCAD #963: Librecad crashing win7 64bit

David87150, 2018-03-24:
Hello
I have just installed Librecad after a few hours work it started crashing and recrashing (after restart, and reinstaillation) i checked auto save set for 5mins but the crash is random, though it gives an indication its about to happen as if something else is happening at the same time. My pc E8400 3ghz runs using normally 20% ram in use, and normally without problems, I have run CCleaner.
Version: 2.1.3
Compiler: GNU GCC 4.9.1
Compiled on: Sep 23 2016
Qt Version: 5.4.1
Boost Version: 1.60.0
System: Windows 7
I can find no upto date help on this issue other than upgrading to next issue, which I have done and still the same.
ANY ideas?

shawncurry, 2019-05-25:
Yeah, need to figure out some reproducible steps.  Drawings, action at the time, etc.

sand1024, 2026-09-26:
too old version, no info

closed 2026-09-26 by sand1024