LibreCAD/LibreCAD #965: Slow Zoom

benohb, 2018-03-28:
### Expected behavior
What is expected
### Observed behavior
The hard drive operates at every click when using the spline or/and zooming 
Something that is written every time (zooming) to a file /home/user/.config/LibreCAD/LibreCAD.conf
 Make this file read-only  ... problem disappears
### Steps to reproduce or sample file
- New file 
- Spline  
- mouse wheel / click
### Operating System and LibreCAD version info
Linux debian 64 
libc6 2.27

benohb, 2018-03-28:
```
 /sudo fatrace 
librecad(20964): CWO /home/hb/.config/LibreCAD/LibreCAD.conf
librecad(20964): O /home/hb/.config/LibreCAD/LibreCAD.conf.lock
librecad(20964): W /home/hb/.config/LibreCAD/LibreCAD.conf.lock
librecad(20964): O /home/hb/.config/LibreCAD/LibreCAD.conf.Z20964
librecad(20964): W /home/hb/.config/LibreCAD/LibreCAD.conf.Z20964
librecad(20964): CW /home/hb/.config/LibreCAD/LibreCAD.conf
librecad(20964): CW /home/hb/.config/LibreCAD/LibreCAD.conf.lock
librecad(20964): CWO /home/hb/.config/LibreCAD/LibreCAD.conf
librecad(20964): WO /home/hb/.config/LibreCAD/LibreCAD.conf.lock
librecad(20964): O /home/hb/.config/LibreCAD/LibreCAD.conf.B20964
librecad(20964): W /home/hb/.config/LibreCAD/LibreCAD.conf.B20964
librecad(20964): CW /home/hb/.config/LibreCAD/LibreCAD.conf
librecad(20964): CW /home/hb/.config/LibreCAD/LibreCAD.conf.lock (deleted)
```

mpiffault, 2019-03-14:
#859

mpiffault, 2019-03-14:
This due to LibreCad writing current tool options to disk on every action, including zooming. See #859 and fix #1079 .

commit 46458dee (2019-03-17): Merge pull request #1078 from xanderdin/settings_write_skip

fix #859, fix #965, Skip re-saving already saved settings values
closed 2019-03-17 by lordofbikes