LibreCAD/LibreCAD #971: Error DXF x, y, z scale not equal

dennisholt, 2018-04-08:
### Expected behavior
dxf 2007 file compatible with other applications
### Observed behavior
LibreCAD drawing with block inserts saved as dxf 2007 generated Error x, y, z scale not equal when imported into LaserCut 5.3.  
I used text editor to set z scale in each INSERT section of dxf equal to the x scale value (which was equal to the y scale). When fixed dxf file was opened in LibreCAD and re-saved the z scale was again set to zero.  Test edit required each time before importing into LaserCut 5.3.
*** Work around ***
Edit dxf file with test editor to set z scale in each INSERT entity section equal to the x scale value.
In this example x scale (group code 41) = y scale (group code 42)= 1; while z scale (group code 43) = 0.
Using TextWrangler search " 30\n0\n 41\n1\n 42\n1\n 43\n0\n 50\n"
replace all with " 30\n0\n 41\n1\n 42\n1\n 43\n1\n 50\n"
With this edit the dxf file imported without error.

### Steps to reproduce or sample file
The following is an INSERT section from one of my LibreCAD dxf 2007 files:
......................
INSERT
  5
22C
100
AcDbEntity
  8
Cut
  6
CONTINUOUS
 62
    6
370
    5
100
AcDbBlockReference
  2
LongCut2
 10
35.2669219352166
 20
30.00013612147712
 30
0
 41
1
 42
1
 43
0
 50
234
 70
    1
 71
    1
 44
1
 45
1
  0
..............

### Operating System and LibreCAD version info
OS:  Apple macOS Version 10.13.3
LibreCAD:  Version: 2.1.3
     Compiler: Clang 8.0.0
     Compiled on: Sep 22 2016
     Qt Version: 5.6.1
     Boost Version: 1.60.0
     System: OS X 10.13

ayaromenok, 2018-06-23:
it's looks like a LaserCut import error, not a LibreCAD - since it's no Z coordinate at all in 2D drawing and any value should be ignored by importing software. 
writing value, other than 0 in Z coordinate may cause situation, that other importers would like to find 3D value.
in case, is LaserCut can import SVG - you can try to use it instead of DXF

dennisholt, 2018-06-26:
Andrey, Thank you for reviewing this issue.  You are correct about Z coordinates but I am talking about Z scale.  A 2D drawing should have Z coordinates of 0. But the Z scale of 0 is the problem.  X and Y scale are quite often 1.00 almost never 0.0.  I think LibreCAD could set the Z scale equal to the X and Y scale when ever they are equal and to 1.00 otherwise.  This would make more sense in general than setting it to 0.0 all the time.  And some day LibreCAD may grow another dimension.  

The editing of LibreCAD DXF file before importing is a work around if you know about it.  I don't think LibreCAD should purposely prissiest in this incompatibility.  The other end of the communications is proprietary software that will not respond to the customers complaints.

dennisholt, 2018-06-26:
If I can get access to some of the proprietary CAD software alternatives to LibreCAD I will try them and see what they produce in the Insert entry.  Perhaps leaving out the group code 43 and its value is the best answer.  I don't know the DXF standard very well.  Hope we can make LibreCAD the best compatible alternative to proprietary 2D CAD software.

ayaromenok, 2018-06-27:
just check a to Autodesk DXF specification (http://images.autodesk.com/adsk/files/autocad_2012_pdf_dxf-reference_enu.pdf, page 98) it's a 
`43 - Z  scale factor (optional; default = 1)` 

so, possible solution is:
 - don't write Z scale at all, since it's a optional group code;
 - write always `1`  as default value, since LibreCAD don't use this value anyway;
 - write current `0` value with additional dialog, where user can select a option from above (from one side, this will leave already existing DXF files, created with LibreCAD, unmodified, from other side it will make more complicated `Save` action by adding additional GUI Dialog)

@lordofbikes  what do you think about such modification?

ayaromenok, 2018-06-27:
@dennisholt - while it will take some time to implement a fix and test\release it, you may need a first working solution, which is better that just a modify DXF file by hands, 

so you can build LibreCAD from sources as described at https://wiki.librecad.org/index.php?title=Build_from_source#Alternative:_Building_from_Downloaded_Source_Code

and modify file 
`LibreCAD/libraries/libdxfrw/src/libdxfrw.cpp:1135` > `bool dxfRW::writeInsert(DRW_Insert *ent)` method  >> `writer->writeDouble(43, ent->zscale);` to writing 1 or not write Z scale at all 

well, it's a hack, but will take a few minutes to fix\rebuild, and you feedback will be welcome.

egilkv, 2021-11-24:
FYI FreeCAD has a similar issue with Z scale of zero, and chokes. https://forum.freecadweb.org/viewtopic.php?f=3&t=63975

I must admit I am not in the position to know whether Z scale of zero is compliant with the spec or not, but it definitely does seem like it creates some real-world problems. Not outputting any Z scale at all might possibly be the cleanest way of dealing with this. 

A patch that make LibreCAD maintain a Zscale of 1 can be found here: https://tracker.freecadweb.org/view.php?id=0004791