LibreCAD/LibreCAD #976: The command line language does not match the setting.

Litanis, 2018-04-22:
### Expected behavior 1
GUI Language: English
Command Language: Russian русский

### Observed behavior 1
GUI Language: English
Command Language: English

In Application Preferences: 
GUI Language: English
Command Language: Russian русский

Command line:
дуга
Unknown command: дуга

### Steps to reproduce 1
1. Application Preferences
2. GUI Language: English; Command Language: Russian русский
3. Restart the program
4. Entering the command line "дуга"
5. In the field of the entered commands appears the message in English "Unknown command: дуга"

If you press the "Tab" button at the flashing cursor on the command line, a list of commands appears in English.

### Expected behavior 2
GUI Language: Russian русский
Command Language: English

### Observed behavior 2
GUI Language: Russian русский
Command Language: Russian русский

In Application Preferences: 
GUI Language: Russian русский
Command Language: English

Command line:
arc
Неизвестная команда: arc

### Steps to reproduce 2
1. Application Preferences
2. GUI Language: Russian русский; Command Language: English
3. Restart the program
4. Entering the command line "arc"
5. In the field of the entered commands there is a message in Russian "Unknown command: arc"

If you press the "Tab" button in the command line with a blinking cursor, a list of commands appears in Russian.

### Operating System and LibreCAD version info
Version: 2.1.3
Compiler: GNU GCC 4.9.1
Compiled on: Sep 23 2016
Qt Version: 5.4.1
Boost Version: 1.60.0
System: Windows (Windows 10)

shawncurry, 2019-05-24:
Confirmed.  Choosing a different command language appears to have no effect; it always uses the GUI language.

However it begs the question, is this feature really needed?  It probably wouldn't work very well for non Latin combinations.  Does anybody really want this?

shawncurry, 2019-05-24:
The feature appears to be a carryover from qcad that was never implemented.

fa201, 2019-05-24:
On my old release it works for French (ligne and cercle works).
Version : 2.1.2
Compiler: GNU GCC 6.2.0
Compilé sur : Sep 17 2016
Qt Version: 5.9.5
Boost Version: 1.62.0
System: Ubuntu 18.04.2 LTS

shawncurry, 2019-05-25:
Hi Fabrice,

Do you also have your UI language set to French?  The combinations that aren't working are when the command language is different from the UI.  I just tried with the 2.1.2 and 2.1.0 versions available on the GitHub release page and both seem to also have the problem.

If this feature was working correctly, do you think you'd use it?  The "real world" use cases seem limited.  Most of the time, users will have a keyboard that matches their preferred alphabet, so they cannot even choose a command language which uses a different alphabet.

Really, it's the keyboard alphabet that will dictate your available choices for command input.  So the only practical case might be if a user is being forced to use a keyboard that does not match with the language they prefer to read with.  

Further, in such a case, wouldn't it only be the actual command entry you'd want to be different?  You'd still want the captions and feedback messages to be translated in the UI language.  Part of the bug appears to be that the translations of the command keywords do not use an alternate translation context, while these may be the thing you'd want the setting to apply to.

fa201, 2019-05-25:
Ooops. You are right I set the GUI to French. I confirm the bug is reproducible. Sorry.
From a user perspective, I do not see such difference as being an issue. Then I would recommend to have 1 drop list for both GUI and command. This should kill any misunderstanding from the user. Having command in English with non-English command should still be possible through alias ?

shawncurry, 2019-05-26:
> Ooops. You are right I set the GUI to French. I confirm the bug is reproducible. Sorry.

No problem, Fabrice; thank you for your testing and input on this case.  Often we developers so not specify meanings as well as we should.  I'm sure you are familiar with this problem.  Haha

It wouldn't be very hard problem to solve.  The application currently creates 3 contexts:  for the UI, the plugins, and for Qt.  We'd need another for the command language.  But the question is, where exactly to apply the alternate translations?  More needs to be understood about this user story.

After I thought about it some more, I did identify a group of users who might often find themselves in this situation:  developers.  It could come in handy for testing perhaps, however an even better approach might be to code some automated tests for this purpose.

Earlopain, 2022-11-10:
Fixing this would be much appreciated.

Translatable commands are in your GUI language, while missing ones are in english. I can type `linie` (line), but must use `mirror`. I would love to contribute translations, but the translation server is still gone.

I wouldn't mind the commands being available in english only. It certainly seems better than having them mixed.

closed 2026-09-04 by dxli