LibreCAD/LibreCAD #978: With "open" menu option - always empty DXFs

CED34, 2018-05-01:
### Expected behavior

See drawing when i open a DXF. :)

### Observed behavior

Empty drawing on DXF load.
I had same problem with 2.1.3, so i updated, but same issue with last version.

When I open librecad and use open in the menu, i have an empty drawing. (autozoom tried ;) ).
Note that in this case, it seems to see modifications as it asks me to save before exiting...

When I open a DXF (made by librecad) by double click on explorer, it works.
When I open using recent files, it works.
I will use this last option to be able to open 2 files in the same instance for copy-paste purpose.

### Steps to reproduce or sample file

Using open menu option on a DXF file (not tried on other file types).
I use http://forum.librecad.org/file/n5713719/test2.dxf has testing file to be sure it's not my-files related.

### Operating System and LibreCAD version info

OS: Windows 10 64bits home (french)
IntelCore I5-4570 (4 cores)
Mem 16Go

LibreCad:
Version: 2.2.0-rc1
Compiler: GNU GCC 4.9.1
Compiled on: Feb 16 2018
Qt Version: 5.4.1
Boost Version: 1.53.0
System: Windows

Knaldgas, 2018-06-28:
### Confirmed
Version: 2.1.3
Compiler: GNU GCC 7.3.0
Compiled on: Mar 24 2018
Qt Version: 5.10.1
Boost Version: 1.62.0
System: Debian GNU/Linux buster/sid

lordofbikes, 2019-03-22:
By accident I found this:
[image]
The default for _Files of type_ is _Drawing Exchange (*.dxf)_.
With _Drawing Exchange_ type files open well, but not with _QCad 1.x file_!
The _Files of type_ selection is stored in settings, so when the type in open file dialog was changed by accident, empty drawings are the result.
Using recent files list, the type detection returns _Drawing Exchange_ and the file opens well.

I can't say yet, why _QCad 1.x files_ results in an empty drawing.
We have to check if _QCad 1.x_ was that different or if there is an issue with this import filter.

shawncurry, 2019-05-24:
Looking at rs_filterdxf1.cpp, the import code appears to be completely commented out, and the export was never implemented.  QCAD 1.x examples would need to be located for inspection and testing in order to implement this.

I recommend discontinuing this feature.  It would be of limited value even if implemented, and qcad likely supports such a conversion anyhow if it was ever needed.

If the file is opened as "Drawing Exchange (*.dxf)", it opens properly:
[image]

melwyncarlo, 2021-11-19:
I don't encounter these problems in both v2.1.3 and v2.2. Has this been fixed?

Knaldgas, 2022-03-24:
Still a problem in v2.1.3 on my machine.
Version: 2.1.3
Compiler: GNU GCC 11.2.0
Compiled on: Jan 30 2022
Qt Version: 5.15.2
Boost Version: 1.74.0
System: Debian GNU/Linux bookworm/sid

Since I do not use explorer, my only option is to Open the file (blank), close it, and then reopen it through recent files.
Quite annoying actually...

telephrag, 2025-11-10:
In case any fix was implemented in code, this issue reappeared for me. I have to fiddle with various rituals to continue working on the drawing.