LibreCAD/LibreCAD #979: Program stops responding consistently when showing a specific construction layer

scrain777, 2018-05-02:
### Expected behavior
I expect the program to remain responsive when a dashed construction line from 2,98 to 1.999999999999993,-6 is rendered off-screen.

### Observed behavior
The program becomes non-responsive when panning, zooming, changing layer visibility or undeleting results in the specified line being potentially visible but off-screen.

### Steps to reproduce or sample file

1. Load the SampleCrash2.dxf file.
[SampleCrash.zip](https://github.com/LibreCAD/LibreCAD/files/1965622/SampleCrash.zip)

or recreate it with:
1. Start LibreCAD.
2. Create a construction layer (essential) and set the line type to dashed or dotted (essential).
3. Type the commands:
line
2,98
1.999999999999993,-6

Then, any of the following will cause the program to freeze:
1. Pan or zoom so that the vertical line goes off the left or right edge of the screen.

Or:
1. Make the layer invisible.
2. Scroll so the line would be off-screen.
3. Make the layer visible.

Or:

1. Delete the vertical line.
2. Scroll so the line would be off-screen.
3. Undo.

### Work Arounds:
If the line type of the layer is changed to continuous the problem goes away. If the layer is changed to not a construction layer the problem goes away. If the layer has visibility disabled, the problem goes away. If this single construction line is removed from the complete drawing, the problem goes away.

### Operating System and LibreCAD version info
Windows 10 64 bit with 16GB ram. 

This problem was originally experienced on version 2.1.3. I upgraded to the release candidate and was able to recreate the problem in the new version.
 
Version: 2.2.0-rc1
Compiler: GNU GCC 4.9.1
Compiled on: Feb 16 2018
Qt Version: 5.4.1
Boost Version: 1.53.0
System: Windows

lordofbikes, 2018-05-02:
There are a couple of similar reports, summarized in PR #915.
A fix was committed in March, your version is build in February, so when you install the latest nightly build from [SourceForge](https://sourceforge.net/projects/librecad/files/Windows/NightlyBuilds/master/), this problem may be solved.
The latest nightly is LibreCAD-Installer-2.2.0-rc1-21-g3b5e31af.exe, 21 commits ahead of your installed version, so you don't have to expect more instability from this version.

lordofbikes, 2020-04-12:
This was fixed with 7bb66bfa10f6f38914b0efa06afaf3b76aa99917, merged with edf15fe9c2df522d8fe97210681663fb0d937a38

closed 2020-04-12 by lordofbikes