LibreCAD/LibreCAD #980: Layer list panel buggy in Ubuntu MATE 18.04 - Icon columns take too much space

j-von, 2018-05-03:
### Expected behavior
Icon column widths being the width of the icon plus a few pixels.
### Observed behavior
Icon columns take most of the space and widths cannot be adjusted.

### Steps to reproduce or sample file
- Install Ubuntu MATE 18.04
- Choose the Pantheon Panel layout in MATE Tweaks
- Install Software Manager 3.28.1
- Install Librecad 2.1.2 through the Software Manager
- Open Librecad 2.1.2

Watch the Layer List Panel

[image]

### Operating System and LibreCAD version info
Ubuntu MATE 18.04 and LibreCAD 2.1.2-1

jgrocha, 2018-05-07:
On my standard (out of the box) gnome based Ubuntu 18.04, the same happens.

[image]

Ubuntu comes with Qt 5.9.5. It might be a Qt issue.

**LibreCAD**
Version: 2.1.2
Compiler: GNU GCC 6.2.0
Compiled on: Sep 17 2016
Qt Version: 5.9.5
Boost Version: 1.62.0
System: Ubuntu 18.04 LTS

lordofbikes, 2018-05-08:
It works fine on 17.10, with Qt 5.9.1.
I build 2.1.3 and latest 2.2.0-rc1 on 18.04, it's all the same as above.
That supports the theory of a Qt issue.
Does anybody know other Qt5 software to verify this on 18.04?

ayaromenok, 2018-06-14:
can it be https://bugreports.qt.io/browse/QTBUG-61244 -  'Fixed centering of some items in layouts', which was fixed in 5.9.3?

spud1259, 2018-09-23:
I have this problem with the layer icon taking up too much space too.  
I just loaded the OS:
Version: 2.1.2
Compiler: GNU GCC 6.2.0
Compiled on: Sep 17 2016
Qt Version: 5.9.5
Boost Version: 1.62.0
System: Ubuntu 18.04.1 LTS

I'd like to be able to modify the columns (add/delete/reorder).  For example: I don't want to see the line color in the layer list.
I also don't need the "shows on print" feature.  I have always changed the visibility before I printed a drawing.

spud1259, 2018-10-09:
Version: 2.1.2
Compiler: GNU GCC 6.2.0
Compiled on: Sep 17 2016
Qt Version: 5.9.5
Boost Version: 1.62.0
System: Ubuntu 18.04.1 LTS

The columns in the layer list window should be configurable as the whether they are visible or not.  

I read the blurb from a developer about construction lines.  First, I found it condescending.  He seems to have needed construction lines in his previous work.  They've always been there, they are on the properties pop-up in previous versions of LibreCAD.

Having a column to "include while printing" is a waste of space for me.  If I don't want it in my prints, then I turn off the visibilty (eye icon) and print... that's always worked in the previous versions.

We all work differently and the UI shouldn't be based one developers whacked out view of what's right or wrong.  In short, make if configurable!

closed 2018-10-10 by lordofbikes

eslavko, 2019-08-18:
With xubuntu 18.04 I have similar problem.
even the icons are werid
[image]

Version: 2.2.0-alpha
Compiler: GNU GCC 7.3.0
Compiled on: Jun 18 2018
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.3 LTS

xanderdin, 2019-08-18:
@eslavko, your LibreCAD binary is compiled on Jun 18 2018. This issue was fixed and closed on Oct 11 2018. Please, try newer LibreCAD version. Hope it helps.

eslavko, 2019-08-18:
that I get with lattest ppa! (daily)

eslavko, 2019-08-18:
https://launchpad.net/~librecad-dev/+archive/ubuntu/librecad-daily

xanderdin, 2019-08-18:
@eslavko, yep, and it is more than year old. I would only suggest you compiling it yourself. It shouldn't be that hard on Ubuntu. Just follow the steps described in https://wiki.librecad.org/index.php?title=Build_from_source#Building_LibreCAD_2.0_on_Linux

eslavko, 2019-08-18:
I tried that but fail. Seems that some dependencies are still not met.
But now I wonder if ppa-daily are not maintained then something is wrong with librecad. 
Is it reach end of life? better to see for some alternatives?

fa201, 2019-08-18:
You have a more up-to-date build instruction at : https://librecad.readthedocs.io/en/latest/appx/build.html#building-on-linux.
It works for my Xubuntu 18.04 LTS.

eslavko, 2019-08-18:
That's works...
but still wonder why ppa-daily is so old...

xanderdin, 2019-08-19:
@eslavko 

Look, every open source project is like a campfire. You can find one, come to it, have light, warmth and even meal. If you find it burning not that strong as you think it should, you are always free to either go and find another one, or think and decide to yourself what can you do to help keeping this fire burning. Paper, dry grass, tinder, kindling, tree bark, twigs, sticks, any other softwood, hardwood, everything can add to its burning.

I mean asking maintainers and developers a question like 'should I go and find something alternative because it is something wrong with your project' sounds like they owe you something and have to please you in other to keep you. Such questions most probably stay unanswered.

Think and ask how can you help and make those ppa released daily. Keep the fire burning! :fire:

eslavko, 2019-08-19:
Wow.
I didn't like to be mean... Just wonder why nice program slowly die.
With my knowledge of programming is it not possible to be maintainer.... 
But can provide some log for fire.

commit 8d8c1545 (2018-10-10): fixed #980, column width in layer list too big since Ubuntu 18.04