LibreCAD/LibreCAD #981: [2.2.0 Release Candidate] Negative numbers are not accepted for Copy/Move input points

flash3780, 2018-05-05:
### Expected behavior
Copy/ Move > First point (0,0) > Second point (-1,0) > Launches Copy/Move Window

### Observed behavior
Copy/ Move > First point (0,0) > Second point (-1,0) > No response with negative input

As a workaround you can flip the points, but it is inconvenient. Prior versions did not have this issue.

### Operating System and LibreCAD version info
MS Windows 10
2.2.0 Release Candidate

lordofbikes, 2018-05-06:
Recently discussed in the forum: http://forum.librecad.org/Problem-with-relative-addressing-td5715903.html
There's no solution so far, assumption is a muparser issue, but it's still work in progress.

vwren, 2018-06-06:
A workaround is to put a mathematical expression in the box, such as " 0-1,0" which it accepts.  It seems to be parsing the minus sign as a mathematical expression with a missing operand.  Supply the operand (zero) and it's happy.

Jim-2249, 2018-10-05:
Same issue with the Lengthen command. "-1" lengthens a line by 1 unit. "0-1" shortens by 1 unit.

linuxguy123, 2019-02-09:
This is totally reproducible on my system.

Command: li (line)
0,0
6,0
li
Command: li (line)
-2,6
Expression Syntax Error

Details:

Version: 2.2.0-alpha
Compiler: GNU GCC 8.1.1
Compiled on: Jul 23 2018
Qt Version: 5.11.3
Boost Version: 1.66.0
System: Fedora 29 (Workstation Edition)

$ uname -a
Linux Brix 4.20.6-200.fc29.x86_64 #1 SMP Thu Jan 31 15:50:43 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

maxcel2, 2019-04-04:
Version: 2.2.0-alpha
Compiler: GNU GCC 7.3.0
Compiled on: Jun 18 2018
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Linux Mint 19.1

The bug is far worse than simply entering '0-10'. 
If I click on Properties for any entity which has a negative co-ordinate, then close the properties by clicking OK, every negative value is set to zero and the entity moves to be anchored on x0, y0 or both depending on whether x, y or both had negative values. The only way out is to inset 0- in front of every negative value before clicking OK - every time Properties is used!! This is a show stopper for me.
This is a serious and major bug which has only just occurred. Up until today I have never experienced this problem.
Is it possible to download an earlier version of librecad which does not have this bug?

patricktokeeffe, 2019-05-01:
This is a total deal breaker. 

With 2.2.0-rc1 (Windows 7) I can't even generate a circle: 
[image]

My setup uses 'inch' for units and English for GUI & command.

maxcel2, 2019-05-02:
Is anybody still maintaining librecad?
This thread has been open for a year now and I cannot see any response from developers.

lordofbikes, 2019-05-02:
This issue is still not reproducible for me, so it's hard to battle this.
Only work around for now is still to enter ***0**-0.1* for negative values.

vwren, 2019-05-02:
Out of curiosity, are other end-users working with 2.2.0-RC1 also unable to replicate this error?  I'm wondering if there's something in the development environment that is "fixing" this.  I have verified the issue that Maxcel2 commented April 4, 2019 on my own setup.

xanderdin, 2019-05-02:
@vwren, I'm not having this issue either.

fa201, 2019-05-02:
I succeeded to draw lines with negative values  for coordinates with 2 versions :
1/ Version: 2.2.0-alpha from Ubuntu PPA as far as I remember
Compiler: GNU GCC 7.3.0
Compiled on: Jun 18 2018
Qt Version: 5.9.5
Boost Version: 1.65.1
System: XUbuntu 18.04.2 LTS

2/ Version: 2.2.0-rc1-111-g46458dee build by myself from Github on April 16th 2019.
Compiler: GNU GCC 7.3.0
Compiled on: Apr 16 2019
Qt Version: 5.9.5
Boost Version: 1.65.1
System: XUbuntu 18.04.2 LTS

maxcel2, 2019-05-02:
After the problem arose, I went back to the Linux Mint (Ubuntu) repository version 2.0.9 with the old style menus. Today, I tried unsuccessfully to install 2.1.3, so I did an apt remove librecad, then added the librecad-daily ppa, updated and installed librecad. 
Strangely, according to Synaptic package manager the version now installed is 2.1.1+20170531.438-0~ubuntu16.04.1, yet the librecad 'Help -> About' says that it is version 2.2.0-alpha. This version, whether it be 2.1.1 or 2.2.0-alpha does not exhibit the negative numbers problem.

xanderdin, 2019-05-02:
I've just freshly installed Linux Mint 19.1 in VirtualBox. Applied all updates to the system, cloned LibreCAD from GitHub and compiled it. No issue with negative numbers. The only strange thing is the About dialog is lacking version info. But no issue with negative numbers appeared.

Version: 
Compiler: GNU GCC 7.4.0
Compiled on: May  2 2019
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Linux Mint 19.1

patricktokeeffe, 2019-05-06:
I no longer think this is parser related. I'm trying to reproduce the issue in a new document today but negative numbers are handled just fine. fwiw I'm also now able to type a space character into the command bar and previously that was interpreted as an enter key.

Jim-2249, 2019-05-06:
patricktokeeffe, Did you change anything?

shawncurry, 2019-05-12:
Confirmed on Win10, Qt 5.12.3/MinGW 7.3 + Visual Studio 2017.  Looks bad.  It works in 2.1.3. 

It's on my radar now.  This is probably going to be an important one for us.

Jim-2249, 2019-05-12:
Another, likely related problem:
On a new page with a fresh start. Draw a two point line from 0,0 to (12-10)/2,0 draws a line from 0,0 to 7,0 - not 1,0. Drawing from 0,0 to (2-1)/2,0 draws a line from 0,0 to 1.5,0  Also moving a (0,0 - 0,1) line moves it to (7,0 - 7,1)

shawncurry, 2019-05-12:
I tried updating the "patched" muparser with the latest headers and sources.  I simply copied over the new files and added "DEFINES += MUPARSER_STATIC" in muparser.pro.  

Same result.  I also tried building _UNICODE vs not unicode.  I did not investigate using an "external" muparser because I'm beginning to suspect it's not the problem.

I logged qg_commandwidget.cpp and found a batch of commits authored by ravas begining June 2016.  I discovered a couple items which seem to involve behaviors discussed on this thread:

> Also, fwiw, in 2.2.0-rc1 the space bar was interpreted as an [Enter] press for me.

> fwiw I'm also now able to type a space character into the command bar and previously that was interpreted as an enter key.

#764 
Possibly #745 (enter or space toggle snap mode)

I personally feel this is a bad change.  When entering a point, most users will intuitively key a space after the comma.  2.1.3 allows the space.

I also noticed that in 2.2, circle mode requires a point-on-circle, where 2.1 took a radius.  However this was consistent with [the documentation](https://wiki.librecad.org/index.php/Commands).  While point-on-circle is the easiest when you're using the mouse, I think the easiest way for command input is point, radius; like it was before.  (There wasn't some sort of "command streamlining" that happened around that time, was there?)

Also, there should be more circle modes.  "cr" seems like a good choice for "circle, radius".

shawncurry, 2019-05-12:
I've found the cause.  The problematic code seems to be related to handling of the ' (foot) and " (inch) imperial notations.  It attempts to do some tricky editing of the command string in order to automatically convert for example 20'2 ==> (20*12+2).  There should probably be some earlier decision making in the pipeline about whether or to apply these conversions.

The commit was made on 10 May 2017 by webmite to /src/lib/math/rs_math.cpp.  

`void RS_Math::imperialTranslate(QString& str) {

    if (str.isEmpty())
       return;
    // put brackets around everything first
    str = "(" + str + ")";
    replaceAll(str,"+",")+(");
    replaceAll(str,"-",")-(");
    // convert foot shortform
    replaceAll(str,"\'","*12+");
    // convert inch shortform
    replaceAll(str,"\"","+");
    // fix for inch with no fraction component
    replaceAll(str,"+)",")");  // -- cleanup
}`

This function ends up converting for example -100 ==> ()-(100), causing the eval function to fail.

shawncurry, 2019-05-12:
This does not appear to be a documented feature.  I'd recommend pulling the plug on this commit.

If the foot conversion is a desired feature, a much better approach would be to actually pre-evaluate the terms.  This string edit approach looks highly bug-prone.

vwren, 2019-05-12:
> If the foot conversion is a desired feature, a much better approach would be to actually pre-evaluate the terms. This string edit approach looks highly bug-prone.

Well done, finding this!  At the very least, it should not be attempting to create an equation for feet where no single-quote foot shorthand exists in the string!  The way it's written would only work for a subset of input cases.

shawncurry, 2019-05-12:
The above pull request reverses the aforementioned commit.  The code also appeared to suffer from unicode string handling issues.

I also opened Issue #1098 Add a preference to allow whitespace in command interface.

shawncurry, 2019-05-13:
Pull Request #1099:
[image]

closed 2019-05-14 by lordofbikes

jsdevel, 2021-01-30:
This is still present on version 2.2.0-alpha.  Using Fedora.  The `0-0.5` format works for specifying points.

lordofbikes, 2021-01-30:
2.2.0-alpha was tagged in 2016, with the release of the first 2.1.0 release.
The fix for this issue was applied in May 2019.
So what is the exact version you have? 
Help -> About, post *Version* and *Compiled on* at least.

PR #1097 (2019-05-14): - fix #981 [2.2.0 Release Candidate] Negative numbers are not accepte…
- fix #981 [2.2.0 Release Candidate] Negative numbers are not accepted for Copy/Move input points
- reverses a commit authored by webmite from 10 May 2017:  Update rs_math.cpp eval fn to allow Imperial shorthand
- Issue #878, Pull Request #885
- this string editing approach appears to be bug-prone; a better solution would be to apply some pre-evaluation of the terms
- code appears to also suffer from incorrect unicode handling.
commit 1e50c8b4 (2019-05-14): Merge pull request #1097 from shawncurry/bugfix/Issue981

- fix #981 [2.2.0 Release Candidate] Negative numbers are not accepte…
commit 7ac6be61 (2019-05-14): - fix #981 [2.2.0 Release Candidate] Negative numbers are not accepted for Copy/Move input points
- reverses a commit authored by webmite from 10 May 2017:  Update rs_math.cpp eval fn to allow Imperial shorthand
- Issue #878, Pull Request #885
- this string editing approach appears to be bug-prone; a better solution would be to apply some pre-evaluation of the terms