LibreCAD/LibreCAD #983: LFF-font format specification

Symbian9, 2018-05-14:
Add full specification on LFF-font format on LibreCAD wiki, as for now there NO (or I don't know where it is)

- http://wiki.librecad.org/index.php?title=Fonts
  - http://github.com/LibreCAD/LibreCAD/tree/master/librecad/support/fonts
  - http://forum.librecad.org/Re-LFF-specification-td5158408.html

There is old (not updated since 2012)...

- http://forum.librecad.org/Where-I-can-find-LFF-specification-arcs-td5158044.html
  - https://github.com/Rallaz/LibreCAD/wiki/lff-definition

... but official full LFF Specification needed.

### REFERENCE

- http://librepcb.org/blog/2018-04-21_librepcb_meets_fontobene/
  - https://github.com/fontobene/fontobene/blob/master/SPECIFICATION.md
  - https://github.com/fontobene/fontobene-converters/tree/master/lff

shawncurry, 2019-05-24:
I have copied rallaz' LFF description to LibreCAD's GitHub wiki:  https://github.com/LibreCAD/LibreCAD/wiki/LFF-Font-Format

While these notes could probably be made a bit more readable, they are technically complete.  The representation is effectively identical to a dxf polyline.  Points are given in a semicolon-delimited list, and arcs are defined in sagitta notation ("bulge").  The format of the file generally follows the AutoDesk shapefile (SHP) format as used for SHX fonts.

[https://en.wikipedia.org/wiki/Sagitta_(geometry)](Sagitta (geometry))
[http://docs.autodesk.com/ACD/2011/ENU/filesACG/WS73099cc142f4875513fb5cd10c4aa30d6b-7f3c.htm](SHP (SHX) font specification)

My coworker was able to write a program in a few hours based on these notes, which he used to convert some homegrown fonts of ours into the LFF format.

[image]

app4soft, 2024-08-31:
> LibreCAD's GitHub wiki: https://github.com/LibreCAD/LibreCAD/wiki/LFF-Font-Format

@dxli This issue could be closed as solved.

dbrgn, 2024-09-01:
(Side note: Probably not relevant for LibreCAD itself, but maybe for other interested people: Given the lack of a formal specification for LFF and the resulting inconsistencies in font files, when we looked for a stroke font format for LibrePCB, we wrote our own specification - strongly inspired by LFF - called FontoBene. https://github.com/fontobene/fontobene/)

app4soft, 2024-09-01:
> Probably not relevant for LibreCAD itself, but maybe for other interested people

*FontoBene* is mentioned on LFF specification page:

- https://github.com/LibreCAD/LibreCAD/wiki/LFF-Font-Format#fontobene-fonts