LibreCAD/LibreCAD #985: Crash when doing rotate - undo - rotate

Daniel-Trevitz, 2018-05-30:
### Expected behavior
not segfault

### Observed behavior
segfaults in void RS_Modification::deselectOriginals(bool remove on line 2043 if (e->isSelected()) {

### Steps to reproduce or sample file
Create a line
Rotate the line by 1 deg deleting the original (I think any amount will do)
Undo with Ctrl-Z
Select and Rotate the line

### Operating System and LibreCAD version info
4.16.0-1-amd64 #1 SMP Debian 4.16.5-1 (2018-04-29) x86_64 GNU/Linux
Librecad trunk

Daniel-Trevitz, 2018-05-30:
this->handleUndo and remove is true at the point of segfault

Daniel-Trevitz, 2018-05-30:
Full backtrace attached
[bt.txt](https://github.com/LibreCAD/LibreCAD/files/2053691/bt.txt)

lordofbikes, 2018-10-11:
Possibly related, at least similar to #993

lordofbikes, 2018-10-20:
My latest fixes for #993 and #1002 changed this behavior too.
No segfault anymore, but the undo buffer is somehow messed up after steps to reproduce.
The first created line couldn't be undone after second rotate.
After second rotate undo till the undo button is disabled, but the line is still there.
After the first rotate it works as expected.