LibreCAD/LibreCAD #988: libdxfrw - crash opening some dwg files

arturredzko, 2018-06-10:
I cannot open these dwg files. The library fails to decode objects handles map correctly generating invalid handle values and object offset locations what leads to corrupted read. Here are samples :

[2000.zip](https://github.com/LibreCAD/LibreCAD/files/2087750/2000.zip)

Are there active dwg reader developers yet to solve this problem?

arturredzko, 2018-06-10:
ok, I found the solution. ModularChar reading can use more than 4 bytes, up to 8 for long handle address.

JackieChiles, 2019-03-03:
I experienced this issue.

Crash dump: https://drive.google.com/open?id=1B4EelDFlTYejUQgQPBsryd43mm1WlB3m

DWG file: https://drive.google.com/open?id=1z0MZCC2HW0b-GzphLbTe5wf6RvLaRtCd

Any ideas?

shawncurry, 2019-05-24:
I can confirm the crash.  This file displays correctly using a different dwg library.  I exported it as a dxf, then loaded that, and all of the text ("Note: this drawing contains an artistic interpretation" etc) got stripped out because the font was not an LFF.  I suspect there is an unhandled exception being thrown for this reason.

shawncurry, 2019-05-24:
Please see feature request #1105 "Improved support for TTF and SHX fonts" for a discussion of a possible solution to this problem.  Your input would be appreciated.

lordofbikes, 2022-11-04:
With the modifications in https://github.com/LibreCAD/libdxfrw the files in question still do not open, but the application does not crash anymore.
Obviously the files contain features which are not yet supported, which needs further investigation.

vehosah, 2025-04-18:
Any update after three years 😕