LibreCAD/LibreCAD #989: Adjustable MakerCAM SVG Export Resolution

dorkmo, 2018-06-10:
### Expected behavior
  * An exported SVG file that replicates the original DXF

### Observed behavior
  * SVG file with low resolution, resulting in details of the DXF shape missing.

### Steps to reproduce or sample file
1. Render OpenScad file
  * https://github.com/OpenSourceEcology/CEB-Press/blob/master/v1708P/review3.scad
2. Export from openscad as DXF
3. Open with LibreCAD.
4. Export to MakerCAM SVG
  * https://github.com/OpenSourceEcology/CEB-Press/blob/master/v1708P/review3-librecadexp.svg

### Operating System and LibreCAD version info
   * Windows 11
   * 2.1.3

### Other Notes
   * When "Exporting as image" there is a dialoge menu to choose resolution.
   * Is there a difference between an export w/ MakerCAM vs w/ export as image?

lordofbikes, 2018-06-11:
There is no need to ask for resolution in SVG export dialog, because DXF and SVG are vector graphic formats.
The SVG export looks low resolution because of the fixed line width in MakerCAM SVG export.
The factor between drawing dimensions (246 x 48) and line width (1) is sub optimal.
My assumption is, that the OpenSCAD file is build in inch. But OpenSCAD, as far as I know, doesn't have a unit property and so DXF export doesn't care about units too.
LibreCAD assigns files without unit information it's default setting from application preferences, which was mm I assume.
On MakerCAM SVG export the line width is fixed 1mm. I can't say from memory if it is considered on export, when current drawing preferences are set to inch. I don't use both, neither MakerCAM export nor inch.
Your options, as I see it, are to modify the line width in SVG to an appropriate value or scale the drawing in LibreCAD by factor 25.4 before export. Maybe changing the current drawing preferences to inch in LibreCAD works too, but I haven't tested this, it's without warranty :wink:.

Finally, I right now merged an improvement from pull request #977, where you can modify the line width in MakerCAM export. But you have to wait until tomorrow, for the nightly build. Then you can download it from [SourceForge](https://sourceforge.net/projects/librecad/files/Windows/NightlyBuilds/master/). We are in pre release state, so this version is stable, even when build from latest code base.

Basically, I see the lack of details already in the OpenSCAD file. All the circles are octagons there. I know there is a variable in OpenSCAD to increase the iterations of circles and arcs. But you'll never get a real circle from OpenSCAD's DXF export. OpenSCAD is not made for this, the usual way is to make a DXF drawing and import it into OpenSCAD to extrude it linear or rotary.

lordofbikes, 2018-06-12:
You're welcome!
It may take a couple days for the nightly build, the internet connection of the build server passed away on Saturday during a thunderstorm, I'm working on it.

ayaromenok, 2018-06-12:
make a smoke test of @dorkmo files with #977(only rectangles from left, since right part is required included files) - as  @lordofbikes mention, you can decrease value from default 1 mm to something like 0.1mm or even less to get necessary resolution during rasterization\printing\graving\etc. 

btw, I still thinking about adding scale (which can be converted to resolution during rasterization) to SVG export plugin.

lordofbikes, 2018-06-14:
The Windows nightly build server is online again, so the #977 improvement is in the latest 2.2.0-rc1-31-g... build

closed 2018-06-14 by lordofbikes