LibreCAD/LibreCAD #993: Crash when doing Cut, paste, undo till before cut

Daniel-Trevitz, 2018-07-10:
### Expected behavior
Not segfault

### Observed behavior
segfaults in 

### Steps to reproduce or sample file
draw two lines. Cut the lines. Paste the lines.
Cttrl-z till crash 

### Operating System and LibreCAD version info
64bit Debian testing & LibreCAD trunk

Rex-On-GitHub, 2018-08-16:
Bug occurs on my system too.

Version: 2.2.0-rc1-36-g439d75d
Compiler: GNU GCC 5.4.0
Compiled on: Aug  7 2018
Qt Version: 5.5.1
Boost Version: 1.58.0
System: Linux Mint 18.3

lordofbikes, 2018-10-10:
fixed with commit 35c23ac

lordofbikes, 2018-10-11:
I didn't realize, that the fix only moves the segfault until LibreCAD is closed.

1. With the fix applied, open LibreCAD.
2. Draw one 2 Points line.
3. Close LibreCAD without saving.
4. LibreCAD doesn't close cleanly, double free or corrupted size exception with memory dump.

Cross checked with the reverted fix, LibreCAD closes fine.
I think this needs a deeper look at undo/redo cycles.

With the fix at least it doesn't crash immediately and gives chance to save work. 
So I'll keep the fix, but reopen the issue.

Daniel-Trevitz, 2018-10-20:
Yeah, my evaluation lead me to believe that the undo framework is simply broken. But I'm not really sure what is wrong...

lordofbikes, 2018-10-20:
I've found the cause for this, will fix it soon.

commit 589e4f26 (2018-10-20): fixed #993, crashed after cut, paste, undo
closed 2018-10-20 by lordofbikes