LibreCAD/LibreCAD #1000: Librecad freezes when zooming in Ubuntu 18.04 on a specific drawing

rnbdc, 2018-08-01:
### Expected behavior
It should be possible to open the file and zoom in with the mouse wheel.

### Observed behavior
Specifically in Ubuntu 18.04 the program will freeze when zooming in and out a few times with the mouse wheel on this drawing. All versions tested suffer from the same problem: 2.1.2, 2.1.3, 2.2.0-rc1-36-g439d75dc

If only "Auto Zoom" and "Window Zoom" are used it works fine. Also works fine in Microsoft Windows.

### Steps to reproduce or sample file
Open the sample file in Ubuntu 18.04 / Librecad, and zoom in with the mouse wheel. Using the "Zoom In" button also triggers the same problem.

[BancadaCozinha-Vistas.dxf.gz](https://github.com/LibreCAD/LibreCAD/files/2249060/BancadaCozinha-Vistas.dxf.gz)

### Operating System and LibreCAD version info
Version: 2.2.0-rc1-36-g439d75dc
Compiler: GNU GCC 7.3.0
Compiled on: Aug  1 2018
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04 LTS

rnbdc, 2018-08-01:
You might need to zoom in and out a few times to trigger the problem. But the bug is consistent.

Rex-On-GitHub, 2018-08-16:
Bug occurs on my system as well.

Version: 2.2.0-rc1-36-g439d75d
Compiler: GNU GCC 5.4.0
Compiled on: Aug  7 2018
Qt Version: 5.5.1
Boost Version: 1.58.0
System: Linux Mint 18.3

mpiffault, 2019-03-14:
I tried to have a grab on what happens.
In informational debug level, it spits a lot of log at the time of the freeze (about 30Mb in 2 or 3 sec).
I got to spot the point of no return from which it falls into an infinite loop :

```
expected array size=4, got 4
x^4+(0)*x^3+(0)*x^2+(2.59063)*x+(-1.67785)==0
lib/math/rs_math.cpp : quarticSolver : line 665
x^4+(0)*x^2+(2.59063)*x+(-1.67785)==0
lib/math/rs_math.cpp : quarticSolver : line 643
expected array size=4, got 4
x^4+(0)*x^3+(0)*x^2+(4.72573)*x+(-5.58313)==0
lib/math/rs_math.cpp : quarticSolver : line 665
x^4+(0)*x^2+(4.72573)*x+(-5.58313)==0
lib/math/rs_math.cpp : quarticSolver : line 643
expected array size=4, got 4
x^4+(0)*x^3+(0)*x^2+(-inf)*x+(-inf)==0            // first anomaly here, where two values are -inf
lib/math/rs_math.cpp : quarticSolver : line 665
x^4+(0)*x^2+(-inf)*x+(-nan)==0                    // second anomaly : last value is -nan
lib/math/rs_math.cpp : quarticSolver : line 643  |// starting here we just have the same 5 lines, over and over
expected array size=4, got 4                     |
x^4+(0)*x^3+(0)*x^2+(-nan)*x+(nan)==0            |
lib/math/rs_math.cpp : quarticSolver : line 665  |
x^4+(0)*x^2+(-nan)*x+(nan)==0                    |
lib/math/rs_math.cpp : quarticSolver : line 643
expected array size=4, got 4
x^4+(0)*x^3+(0)*x^2+(-nan)*x+(nan)==0
lib/math/rs_math.cpp : quarticSolver : line 665
x^4+(0)*x^2+(-nan)*x+(nan)==0
lib/math/rs_math.cpp : quarticSolver : line 643
expected array size=4, got 4
x^4+(0)*x^3+(0)*x^2+(-nan)*x+(nan)==0
lib/math/rs_math.cpp : quarticSolver : line 665
x^4+(0)*x^2+(-nan)*x+(nan)==0
...
```

Hope it can help.

lordofbikes, 2019-03-15:
@mpiffault which version did you use?

mpiffault, 2019-03-15:
@lordofbikes sorry:
```
Version: 2.2.0-rc1-105-gc2635841
Compiler: GNU GCC 7.3.0
Compiled on: Mar  9 2019
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.2 LTS
```

lordofbikes, 2019-03-15:
No problem @mpiffault, no reason to feel sorry, your efforts are welcome.

I had a slim hope that you tried the referenced 2.2.0-rc1-36-g439d75d from above.
There are some similar issues collected in #915 which seamed to be solved.
I remember, that I had similar output with ```-inf``` and ```-nan``` numbers back then.
So that means that these issues are not or not completely solved yet.

mpiffault, 2019-03-17:
After a bit of debugging, here is what I got :

* The problematic entity is a LC_SplinePoints of id 43048 :
```
this	@0x555557aba710	LC_SplinePoints
	[RS_AtomicEntity]	@0x555557aba710	RS_AtomicEntity
		[RS_Entity]	@0x555557aba710	RS_Entity
			[RS_Undoable]	@0x555557aba710	RS_Undoable
			id	43048	unsigned long
			layer	@0x555556f50430	RS_Layer
			maxV	@0x555557aba748	RS_Vector
			minV	@0x555557aba728	RS_Vector
			parent	@0x55555689d5b0	RS_Graphic
			pen	@0x555557aba778	RS_Pen
			updateEnabled	true	bool
			varList	<0 items>	std::map<QString, QString, std::less<QString>, std::allocator<std::pair<QString const, QString> > >
	data	@0x555557aba7f0	LC_SplinePointsData
		closed	false	bool
		controlPoints	<7 items>	std::vector<RS_Vector, std::allocator<RS_Vector> >
			[0]	@0x555557c91240	RS_Vector
				valid	true	bool
				x	97.0	double
				y	67.0	double
				z	0.0	double
			[1]	@0x555557c91260	RS_Vector
				valid	true	bool
				x	97.0	double
				y	73.0	double
				z	0.0	double
			[2]	@0x555557c91280	RS_Vector
				valid	true	bool
				x	97.0	double
				y	77.0	double
				z	0.0	double
			[3]	@0x555557c912a0	RS_Vector
				valid	true	bool
				x	97.0	double
				y	80.0	double
				z	0.0	double
			[4]	@0x555557c912c0	RS_Vector
				valid	true	bool
				x	100.0	double
				y	98.0	double
				z	0.0	double
			[5]	@0x555557c912e0	RS_Vector
				valid	true	bool
				x	112.0	double
				y	125.0	double
				z	0.0	double
			[6]	@0x555557c91300	RS_Vector
				valid	true	bool
				x	113.0	double
				y	145.0	double
				z	0.0	double
		cut	true	bool
		splinePoints	<0 items>	std::vector<RS_Vector, std::allocator<RS_Vector> >
```

* Stack :
```
1  RS_Math::quarticSolver					rs_math.cpp            653  0x55555574b374 
2  (anonymous namespace)::GetQuadPointAtDist	lc_splinepoints.cpp    215  0x5555556921fb 
3  DrawPatternQuad						lc_splinepoints.cpp    1635 0x55555569c515 
4  LC_SplinePoints::drawPattern				lc_splinepoints.cpp    1785 0x55555569de9a 
5  LC_SplinePoints::draw					lc_splinepoints.cpp    1937 0x55555569fb28 
6  RS_GraphicView::drawEntityPlain			rs_graphicview.cpp     1191 0x555555734189 
7  RS_GraphicView::drawEntity				rs_graphicview.cpp     1145 0x555555733e21 
8  RS_GraphicView::drawEntity				rs_graphicview.cpp     1094 0x5555557338b2 
9  RS_EntityContainer::draw					rs_entitycontainer.cpp 1771 0x5555556530e7 
10 RS_GraphicView::drawEntityPlain			rs_graphicview.cpp     1191 0x555555734189 
11 RS_GraphicView::drawEntity				rs_graphicview.cpp     1145 0x555555733e21 
12 RS_GraphicView::drawEntity				rs_graphicview.cpp     1094 0x5555557338b2 
13 RS_GraphicView::drawLayer2				rs_graphicview.cpp     930  0x5555557332c1 
14 QG_GraphicView::paintEvent				qg_graphicview.cpp     936  0x55555585a784 
```

* Crashing call is at lc_splinepoints.cpp:1635 (3rd line in the stack):
`dt2 = GetQuadPointAtDist(x1, c1, x2, dt1, dCurSegLen);`
The attributes passed taking the following values :
```
x1	@0x7fffffffba20	RS_Vector &
	valid	true	bool
	x	390.60501581260439	double
	y	259.87076227376809	double
	z	0	double
c1	@0x7fffffffba40	RS_Vector &
	valid	true	bool
	x	390.60501581260439	double
	y	230.57560665649362	double
	z	0	double
x2	@0x7fffffffba60	RS_Vector &
	valid	true	bool
	x	390.60501581260439	double
	y	220.81055478406876	double
	z	0	double
dt1	0	double
dCurSegLen	37.768924302788847	double
```

Into `GetQuadPointAtDist` a `dDet` variable is computed as the following :
```
RS_Vector xh1 = (c1 - x1)*2.0;
RS_Vector xh2 = (x2 - c1)*2.0;
RS_Vector xd1 = xh2 - xh1;
RS_Vector xd2 = xh1;

double dx1 = xd1.squared();
double dx2 = xd2.squared();
double dx12 = xd1.x*xd2.x + xd1.y*xd2.y;
double dDet = dx1*dx2 - dx12*dx12;
```
Which result as a value of
```
RS_Vector xh1 = (c1 - x1)*2.0;
RS_Vector xh2 = (x2 - c1)*2.0;
RS_Vector xd1 = xh2 - xh1;
RS_Vector xd2 = xh1;

double dx1 = xd1.squared();
double dx2 = xd2.squared();
double dx12 = xd1.x*xd2.x + xd1.y*xd2.y;
double dDet = dx1*dx2 - dx12*dx12;
```
Which results in a value of `2.3283064365386963e-10` very close but greater than RS_TOLERANCE (1.0e-10).

Then we have :
```
double dA = sqrt(dDet);
double v1 = (dx1*t1 + dx12)/dA;
double dB = 2.0*dx1*sqrt(dx1)*dDist/dDet + LenInt(v1);
```
Here LenInt which is defined as
```
double LenInt(double x)
{
	double y = sqrt(1 + x*x); 
	return log(x + y) + x*y;
}
```
returns -inf, because inside it, y and x are opposites (x being equal to `-69418862.568198338`), and so `log(x + y)` => `log(0)` => `-inf`.

I'm not sure about the role of the `LenInt`, but what is sure is that it will gives `-inf` for any value which is low enough. After some tests, it' below -67108863 (67108863 being 0x3FFFFFF and 67108864 being 0x4000000, I'm not shure for negative representation).

mpiffault, 2019-03-18:
Here is a drawing with the bad spline isolated : [crash-spline.zip](https://github.com/LibreCAD/LibreCAD/files/2980113/crash-spline.zip)

It seams that the problematic part are the first 4 aligned points.

I could reproduce reliably the problem by using the "Spline by points" tool with grid snapping and by placing a first point at [0,0], then one at [0,60], then one at [0,100] and it crashes really short after when moving mouse on any point above (on the x = 0 axis).

I could also reproduce it by trying to align more than 3 points in any direction (horizontally, vertically or diagonally) with the "Spline by points" tool.

fa201, 2019-05-09:
I have the same issue when creating a spline through points. Wheel zoom or zoom pan lead to LibreCAD freeze at the first, second or third spline. 
Version : 2.2.0-rc1-115-gc8c5948b
Compilateur : GNU GCC 7.3.0
Compilé le : Apr 27 2019
Version de Qt : 5.9.5
Version de Boost : 1.65.1
Système: Ubuntu 18.04.2 LTS

dxli, 2023-05-22:
The same as #1610. fixed.

closed 2023-05-22 by dxli