LibreCAD/LibreCAD #1004: Crash by doing mv with delete, undo, mv with delete.

Daniel-Trevitz, 2018-08-09:
### Expected behavior
No crash

### Observed behavior
Crashes

### Steps to reproduce or sample file
Create a line. Dimension the line.
move the dimension with the mv command. delete the origonal with the mv.
undo the mv
move the dimension with the mv command. delete the origonal with the mv.
crashes

### Operating System and LibreCAD version info
Linux Daniel6430 4.16.0-2-amd64 #1 SMP Debian 4.16.16-2 (2018-06-22) x86_64 GNU/Linux
branch master

Daniel-Trevitz, 2018-08-09:
Crash occurs in void RS_Modification::deselectOriginals(bool remove) 
if (e->isSelected()) {  <-- crashes here
    selected = true;
}

e is previously deleted by bool RS_EntityContainer::removeEntity(RS_Entity*),

lordofbikes, 2018-10-28:
This was possibly fixed with 2d8e025, but at least with ed90a19

closed 2018-10-28 by lordofbikes