LibreCAD/LibreCAD #1005: Command Line Commands in Tooltips, Menus

NJDFan, 2018-08-09:
Documentation on how to use the command line commands is pretty buried.  Back when I used AutoCAD regularly, I got pretty used to typing commands and modifiers while mousing for actual locations.

If the toolbar buttons included the command line as a parenthetical, it would mean that at least the tools you're using a lot, you learn the shortcuts for free.  So the tooltip would read "Snap on Endpoints (se)", or under Circle you'd have "2 points (c2)"

HerrSubset, 2019-01-15:
I think this would be a great feature too. This way you don't need the documentation to find out about them.

pgporada, 2019-07-04:
I agree this would help. I am a new LibreCAD user.
Related to #570

unixhex, 2023-03-17:
Yes, please. I am currently considering to use either LibreCad or QCad to create plans for a private renovation / interior design project and considering to use LibreCad (and I like it much more than QCad, which I used in previous projects. SVG export, better looking, ... ). BUT the current QCad has all its commands embedded as mouse over, which makes it much easier to find and remember the command line commands.

fa201, 2023-08-09:
You can also find the shortcuts in the reference manual. Example : https://librecad.readthedocs.io/en/latest/ref/tools.html#line