LibreCAD/LibreCAD #1010: statusbar clipped problem

tiredengineer, 2018-08-25:
Just installed LibreCAD on Win10 64b PRO  on HP Envy i7 3840 x 2160 screen resolution and started a learning tutorial.
The important statusbar information is clipped as shown in the PNG.  I applied the Options->Widget Options->statusbar height->99.  The 99 only seems to effect the left set of figures in the red box, the clipped portion in the right hand red box does not change so I don't know what I may or may not be missing.
The display adapter in the HP Envy is an Intel HD Graphics 620.  I tried entering different screen resolutions, but all that happened is the LibreCAD window rescaled with the clipping in the same proportion.
LibreCAD
Version: 2.1.3
Compiler: GNU GCC 4.9.1
Compiled on: Sep 23 2016
Qt Version: 5.4.1
Boost Version: 1.60.0
System: Windows 
[image]

tiredengineer, 2018-08-25:
On further considering the problem it would seem to me that the "statusheight" value is not being applied to the additional fields of the statusbar, hence the constant clipping due to the high resolution screen of the HP laptop.  As an additional thought--perhaps the limit of 99 for height could be increased as perhaps even higher resolution screens may come into use in the future.

tiredengineer, 2018-08-26:
also tried a Win7 compatibility mode and then also 640x480 display mode. Neither helped.  Still have the same clipped bottom.  I'm not positive, but I don't think I can display grids either.

lordofbikes, 2018-08-27:
I assume that we have two issues here.
As you noted, it seems that the mouse button widget ignores the status bar height.
Other issue is the widgets size at all. I assume you have set 125% or 150% font size in display settings. Then the text needs more space, which seems to be not considered and make it look that compressed.
This is a common issue which may concerns other widgets too.

tiredengineer, 2018-08-27:
Armin,
I'm sorry, but I've never used this software before so I'm not sure I 
understand the "set 125% or 150% font size in display settings". Are you 
saying a setting somewhere in the HP laptop display system settings or a 
setting somewhere in LibreCAD?  The problem I submitted is pretty much  
the defaults that LibreCAD starts with. The only thing I did was search 
the Wiki and find out about the statusbar height mentioned in a previous 
bug report and tried setting it as high as it will go (99) and saw that 
only the coordinates displayed in the lower left corner changed their 
height slightly (but they didn't have a problem to start with) and I 
think the window for them changed a bit also.  But nothing else changed 
so the other items essentially remained out of sight due to the clipping 
and as they seem associated with the statusbar I mistakenly thought they 
would have a common height setting arrangement.

So is this a Windows problem, an HP laptop problem, or a more general 
LibreCAD problem?  Is there anything you want me to try?

Regards,
Charles Patton

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

lordofbikes, 2018-08-27:
Charles,

the font size is a display setting in windows preferences. I assume, that big fonts (more than 100%) is set by default from HP, otherwise on HD laptop displays, even 17", the icon and system text is too small.
So reset this option to 100% will bring other issues with readability.
This is an issue on modern 4k UHD displays too, as standard font size is not readable, even on 28" displays. This is a matter of displays pixel per inch.

In other words, I think there's nothing you can do to solve this for you.
This is a common problem for many other software too and has to be considered in GUI programming.
Sadly, this is often left in the cold. Hopefully somebody takes this challenge now.

melwyncarlo, 2021-11-17:
> As you noted, it seems that the mouse button widget ignores the status bar height. 

It no longer does.

melwyncarlo, 2021-11-17:
I would suggest fixing the status bar widgets' heights as a percentage of the device screen.

Software don't usually allow the user to change their status bar graphics. 
Because its contents are not constant, unexpected graphical changes 
would result in unexpected view behaviour.

After all, a status bar is not meant to be used, but just viewed.

melwyncarlo, 2021-11-19:
I've got another idea:

When the application window's width is below a certain minimum value, 
how about we hide a couple status bar widgets that seem non-essential.

My opinion for essential widgets would be the two co-ordinate position widgets and the mouse widget.

What do you think?

commit eb022396 (2022-06-03): sync statusbar and mouse widget heights, fixes #1010,#1026,#1303,#1304
closed 2022-06-03 by lordofbikes