LibreCAD/LibreCAD #1014: The blockName is garbled when a DXF is Encode With GB18030? I suggest that before loading a dxf, checking the dxf encoding,transform it to UTF8.

CodingGou, 2018-09-18:
### Expected behavior

### Observed behavior

### Steps to reproduce or sample file

### Operating System and LibreCAD version info

lordofbikes, 2018-09-18:
Can you attach a sample file please?

CodingGou, 2018-09-19:
[男长裤.dxf.zip](https://github.com/LibreCAD/LibreCAD/files/2395124/dxf.zip)
like this ?

CodingGou, 2018-09-19:
blockName is totally garbled
[image]

CodingGou, 2018-09-19:
the blockname is totally garbled
[image]

shawncurry, 2019-05-24:
There is currently no way for the user to specify alternate encodings during import.  A workaround would be to export them as UTF-8 with whatever program created the file.

CodingGou, 2019-06-04:
I have solved that problem , before parsing the  dxf file，I use a tool to check the dxf file encoding type ,then convert the encoding type you want.