LibreCAD/LibreCAD #1016: Command Line does not respond to all key presses

gavin-blakeman, 2018-09-30:
### Expected  behavior
For instance when selecting line 2-points, the expected behaviour is that after the first point is entered with the mouse, the second point can be directly typed in autocad style. "@250<15". 

### Observed behavior
The keypress of the "@" is not detected, so typing "@250<15" displays on the screen "250<15" and this gives the incorrect response. If any number key is pressed, "2", then deleted, the CLI works as expected and detects the "@" key when pressed.

### Steps to reproduce or sample file
Select line 2 point
Select first point with mouse
Type "@250<15" and see command line "250<15"

### Operating System and LibreCAD version info
LibreCad V2.1.2 Sep 17, 2016
Ubuntu 18.04.1 LTS

lordofbikes, 2018-10-11:
Thanks for reporting this!

closed 2018-10-11 by lordofbikes

tochimoreno, 2020-04-02:
I reported this as duplicate (#1115) before, I install a new version now but this issue is still present in my machine.
Maybe it is because the way I have to introduce the special char (@). This char isn't in my keyboard so I have to press "ALT GR" key and "Q". The "ALT GR" is a standard key in Spanish keyboard but not in English ones.
If there is a way I can help with this, looking fot key code or something, please let me know.

Thank you 

Version: 2.1.3
Compiler: GNU GCC 8.3.0
Compiled on: May 16 2019
Qt Version: 5.12.4
Boost Version: 1.67.0
System: Ubuntu 19.10

hfuerst, 2023-11-02:
This problem ptill exists !!!
@ is not recognized

LibreCad installed Today:
Version: 2.2.0.2
Compiler: GNU GCC 13.1.0
Erstellt am: Jul 29 2023
Qt Version: 5.12.11
Boost Version: 1.75.0
System: Windows 10 (10.0)

Please Open Issue Again

commit d779df21 (2018-10-11): fixed #1016, '@' key press didn't activate command box