LibreCAD/LibreCAD #1019: Angular dimension not moving during a copy, move, or drag.

labusker, 2018-10-11:
[image]
[image]

labusker, 2018-10-11:
When an angular dimension is used in a drawing and then copied or moved to another position, the angular does not move with the other objects. If I save the file, close it, then reopen it, the angulars are then shown to be moved correctly. This will happen 100% of the time on my windows 10 computer using librecad 2.2.0-rc1. I installed 2.1.3 on my surface computer and angular dimensions move correctly. I then updated to 2.2.0-rc1 and then had the same angular move problem. I uninstalled 2.2.0, then reinstalled 2.1.3 and all worked fine.

closed 2018-10-11 by lordofbikes

lordofbikes, 2018-10-11:
Thanks for reporting this!

commit b8e2a3c4 (2018-10-11): fixed #1019, angular dimensions failed to move, rotate, mirror or scale