LibreCAD/LibreCAD #1020: Menu windows too compacted using a Microsoft Surface Tablet

labusker, 2018-10-13:
### Expected behavior

### Observed behavior
Certain windows are too compacted. These include anything containing a checkmark, layer list,  tool option  toolbar, and properties window. I have entered reasonable values in the Widget Options window to fit in my 12" diagonal tablet which makes my icon sizes workable but other windows to compacted. I can enter the height and font size in the Widget Options window, but not the width. I could use a width tool to fix many windows.
### Steps to reproduce or sample file
See attached pictures.
### Operating System and LibreCAD version info
I am using Windows 10 on a 12" diagonal Surface tablet.
[image]

labusker, 2018-10-13:
I am using Librecad version 2.1.3

lordofbikes, 2018-10-14:
My investigation showed that the Surface may have an UHD display, can you confirm this?
What is the display resolution and font size setting?
Then this is may be related to [#1010](https://github.com/LibreCAD/LibreCAD/issues/1010#issuecomment-416170390)

labusker, 2018-11-01:
My surface has a recommended resolution of 2736 x 1824, which I use.  With this kind of resolution, the text, apps, & other items are too small, so the recommended size is to put them at 200%. By adjusting the resolution to 1680 x 1050 and text size to 125% produced good result. Should I settle for this lower resolution or add a width size to the Widget Option Tool Bar? The attachment is the lower resolution.
[image]