LibreCAD/LibreCAD #1021: Printer and construction lines icons are bigger than their cells in Layer List

xanderdin, 2018-10-14:
This issue is obviously created by the fix https://github.com/LibreCAD/LibreCAD/commit/8d8c154552be592960259f570484c32734f3f73a for the https://github.com/LibreCAD/LibreCAD/issues/980 issue.

Currently it looks like this:

[image]

Before the fix it was looking like this:

[image]

Version: 2.2.0-rc1-50-gd90dbd2b
Compiler: GNU GCC 6.3.0
Compiled on: Oct 14 2018
Qt Version: 5.7.1
Boost Version: 1.62.0
System: Debian GNU/Linux 9 (stretch)
MATE Desktop Environment 1.16.2

lordofbikes, 2018-10-14:
It's correct, the icons was adjusted, because they looked smaller before, on my system. Now it's ok for me.
[image]

The difference in build environments is GCC 6.3 vs. 7.3 and Qt 5.7.1 vs. 5.9.5, Boost shouldn't matter.
I suspect Qt, but because the freeze and lock icons are fine on both systems, we have to check the icon creation process.
Another possible cause is display resolution, I use UHD display with 150DPI setting in Ubuntu 18.04.
What is your display size?

closed 2018-10-14 by lordofbikes

lordofbikes, 2018-10-14:
If this fix doesn't work for you, please report and we will reopen the ticket.

xanderdin, 2018-10-14:
It works. Thank you very much!

lordofbikes, 2018-10-14:
Glad to hear this. Many thanks for reporting! :+1:

commit 0e8927d8 (2018-10-14): fixed #1021, oversized layer list icons for print and construction