LibreCAD/LibreCAD #1024: Multilayer block import makes 'Hide/Show all layers' semifunctional

xanderdin, 2018-10-22:
### Observed behavior
After importing a block with several layers, 'Hide all layers'/'Show all layers'
(and also 'Lock all layers'/'Unlock all layers') actions are switching not all layers.

### Expected behavior
Those actions should always switch all layers.

### Steps to reproduce
1. Open a drawing (or create new).
2. Import another drawing containing several layers.
3. Press 'Hide all layers' button (or 'Lock all layers' button).

This bug is not constant. On some repetitions of the above steps the number of
unaffected layers may differ. Sometimes (very rare), those steps don't produce this
bug.

This bug does not appear after closing (with saving) and reopening the drawing.
Only after inserting new multilayer block.

### Operating System and LibreCAD version info

Version: 2.2.0-rc1-62-g4d559536
Compiler: GNU GCC 6.3.0
Compiled on: Oct 19 2018
Qt Version: 5.7.1
Boost Version: 1.62.0
System: Debian GNU/Linux 9 (stretch)

xanderdin, 2018-11-04:
I found the source of this bug. There is **visibleInLayerList**  boolean field in layer data structure. That field is not set during addition of layers from imported drawing. Actually that field gets a random value.

I found two ways how to fix it but which one is better requires some considerations.

1. One can add the following to the `QG_LayerWidget::layerAdded(RS_Layer*)` method:
```
if (layer)
    layer->visibleInLayerList(true);
```
It is straightforward and fast. But the LayerWidget has a filter input field and if a user uses that field to filter out some layers and tries to import a block with new layers, then new layers would be displayed in the layers list regardless of the filter input. Though in this case user can see imported layers right away, it is rather inconsistent with the filter input.

2. Instead, one can add the following to the same `QG_LayerWidget::layerAdded(RS_Layer*)` method:
```
if (layerList)
    slotUpdateLayerList();
```
This is obviously a little bit slower but it is fully consistent with the filter input.

lordofbikes, 2018-11-07:
The root of the problem is finally the missing initialization in `RS_LayerData` class.
Only two of six `bool` are initialized.

So to solve point 1 I would give them valid defaults in the class:
```
    bool frozen {false};
    bool locked {false};
    bool converted {false};
    bool visibleInLayerList {true};
```
I prefer the C++11 list-initializer `{}` over assignment `=` for default initialization, it's more type safe.
And by touching this file I'd change the other two `bool` initialization to `{}` too.

Concerning 2, I think the place is correct. But `slotUpdateLayerList()` is only for filter concerns, so I would check this way:
```
    if (! matchLayerName->text().isEmpty()) {
        slotUpdateLayerList();
    }
```

`layerList` should never be `nullptr` as layer 0 couldn't be deleted. And calling `slotUpdateLayerList()` is only necessary when a filter is active.
With this additions the `QG_LayerWidget::layerAdded(RS_Layer*)` method is way too much for header definition. Means keep the declaration in the header and move the definition to the cpp file then.

xanderdin, 2018-11-08:
Thank you for the explanation. I got your point. So, to work it way 2 both your additions are required. Your approach is definitely more thoughtful. Would you commit the changes to the master then, please? Thank you once again.

lordofbikes, 2018-11-08:
You're welcome!
Sure, if you don't want to carry off the laurels, I can fix this later :wink:

commit ccf50e3e (2018-11-08): fixed #1024, random LayerList icons issues on multi layer block imports, many thanks to @xanderdin
closed 2018-11-08 by lordofbikes