LibreCAD/LibreCAD #1025: Feature Request: Xrefs

turtle0x1, 2018-10-30:
### Expected behaviour

If you have used AutoCAD you will probably be aware of xref feature it allows you reference multiple external drawings in one "master" drawing, this feature is very useful when you are working with multiple client drawings that change frequently 

### Observed behaviour
Not yet implemented

### Steps to reproduce or sample file
N/A

### Operating System and LibreCAD version info
All

redb-2000, 2024-12-18:
hey guys, thanks for all who have been working on this. is there any progress on having the xref feature in librecad? (to be able to reference another librecad drawing in a drawing). 
I have used this feature extensively in autocad, and to make the switch to librecad, this is the most basic feature i need :/

*not a coder here, but hoping to get some sympathy from coders who can do this

dxli, 2024-12-18:
Hi @redb-2000,

Could you provide detailed description of the feature? We are not users of other car softwares, at least, not assumed to be.

For sure, we can attract more interested developers, with more robust development.

redb-2000, 2024-12-20:
Sure @dxl
External Reference (xref) feature allows one to insert another drawing as a guide/reference. The reference drawing appears like a block - but as is saved as a separate file, and any changes made there are reflected wherever it is being used as a reference (so long the reference file has same name and location).
I find it as a crucial feature when working on architectural drawings - as my plan drawing becomes guide for section drawing, then these become guide for plumbing/electrical/furniture/   drawings.

hope this explanation helps some

youtube ref here : https://www.youtube.com/watch?v=wfsUyR9xp70

Udrakan, 2025-12-04:
Hello, AutoCAD expert here. External references are not only an AutoCAD feature, they are crucial for drafting in any construction software. Let me tell you about how they are used:
- Construction engineering company drafts a layout of the first floor
- HVAC company refers to that drawing and only draws HVAC systems in it
- Electrical company also refers to the drawing and to the HVAC drawing to avoid their systems and drafts plugs
- The original construction company changes the layout for any reason (customer wishes, govt etc.)
- Both HVAC and Electrical see the changes and adjust accordingly

Analogy to real life would be a transparent sheet of paper over a construction drawing.
We only have elements of our profession in our drawings and everyting underlaying it is in an immutable xref.
Great feature is being able to toggle layers that are visible on the xref (but it doesn't touch the xref file, just a view, akin to database view)