LibreCAD/LibreCAD #1026: status bar is resized during showMessage() on Ubuntu 18.04 and later

lordofbikes, 2018-10-30:
### Expected behavior
The status bar height is fixed

### Observed behavior
During showMessage(), e.g. on auto save, the status bar height is reduced.
This makes the layout jump on every auto save cycle.

### Steps to reproduce or sample file
Open a new file
Draw some entities
Wait for auto save

For faster success reduce the auto save timer to minimum, means 1 minute

### Operating System and LibreCAD version info
Ubuntu 18.10
LibreCAD 2.1.2

see forum thread http://forum.librecad.org/cursor-jumps-during-autosave-tp5716571.html for more details

martinvanzijl, 2020-04-26:
The following workaround solves the problem for me on Ubuntu 18.04:

1. Select _Options --> Widget Options_.
2. In the _Statusbar_ section, check the _Height_ box.
3. Enter “55” in the text field.

This set the minimum height of the status bar, so it won't resize below this during an autosave.

melwyncarlo, 2021-11-19:
Kindly look into my pull-request #1410 for the implemented solution.

closed 2022-06-04 by lordofbikes