LibreCAD/LibreCAD #1027: Not Responding

labusker, 2018-11-01:
### Observed behavior
 The document I am sending you contains a lot of lines and when I begin zooming in, the program warns that it is "Not Responding". I have to close and relaunch the program to continue. It also produced this error when I tried to move a number of objects. It progressively got worse as I continued on drawing objects.

### Steps to reproduce or sample file
This error is now produced every time I perform the following steps. I open the file. Its "Grid Status" = 10/100. If I zoom in 3 times with the "Zoom In" button, the program freezes.

### Question
I want to re-install the previous version. Do I have to uninstall my 2.2.0 1st? Will I loose all my settings and Layer List?
I tried to attach my .dxf file here, but could not?

### Operating System and LibreCAD version info
Windows 10, Librecad 2.2.0 - rc1

labusker, 2018-11-01:
I transferred this file to my Surface computer which uses 2.1.3 and still had the program freeze when I tried to zoom in.

labusker, 2018-11-01:
I narrowed down the culprits that are causing the freezing. They are my vertical construction lines (Helper3 Layer). If I remove more than 2/3 thirds of them I am able to zoom in 13 times as compared to 4 times with all of them. If I remove all but two of them, zoom works fine. If I add new lines on the left side, problem shows back up. If I remove the construction layer "#" sign, all will work. I kept trying to narrow the problem further, but was hit and miss.
My only solution for now is to not use construction lines; just use long normal lines.

lordofbikes, 2018-11-01:
Yes, this was one of the reasons, vertical construction lines.
Sadly this issue is not solved yet.

labusker, 2018-11-02:
Vertical Construction Lines did cause crashes in a drawing that has a lot of lines. I solved that problem, but not the crashes. Every 3rd to 5th time I manipulate an item on the drawing, it crashes. The only thing that keeps my sanity is that I save my file every other time I add or change something.

lordofbikes, 2018-11-03:
This could possibly be related with undo/redo issues.
I hopefully fixed them lately (Oct. 28th), so you can try to update to LibreCAD-2.2.0-**rc1-69** or newer.
https://sourceforge.net/projects/librecad/files/Windows/NightlyBuilds/master/

If you didn't use undo with your modifications, the update presumably won't help.
Then you should share the file if possible.

labusker, 2018-11-06:
Today I loaded the latest update to my computer. I turned on all my construction lines and performed a lot of work on my busy drawing. So far, knock on wood, I have not got any freeze-ups nor crashes.
Thanks Librecad Team!
[image]

lordofbikes, 2018-11-07:
I'm glad to hear this.
So I close this issue for now.
Feel free to reopen it, if there are still any problems.

closed 2018-11-07 by lordofbikes