LibreCAD/LibreCAD #1035: command line enhancement

WolfgangFahl, 2018-11-06:
### Expected behavior
`pl (0,0 5000,0 5000,1300 0,1300) `
should work

### Observed behavior
pl interactively asks for parameters

### Steps to reproduce or sample file
https://wiki.librecad.org/index.php?title=Starting_to_draw

### Operating System and LibreCAD version info
Mac OS High Sierra / 2.1.3

lordofbikes, 2018-11-07:
Using spaces to separate multiple commands doesn't work, because SPACE was implemented for AutoCAD users, to react like ENTER (see #764).
2.1.3 already has the multiple command feature, see #907.
Beside the options paste and file, shown in the #907 answer, you can also type the commands into the command line and push ENTER.

So the correct syntax for your command is:
`pl;0,0;5000,0;5000,1300;0,1300`
or
`pl;0,0;5000,0;5000,1300;close`
and then ENTER to run all the commands

closed 2018-11-07 by lordofbikes

WolfgangFahl, 2018-11-07:
Thank you for the feedback I tried this out and couldn't get it to work on my 2.1.3 version on MacOS.
I did a cut&paste of 
`pl;0,0;5000,0;5000,1300;close`
into the command line hitting enter and got no visible result but also no error message.
Doing the step by step approach works.