LibreCAD/LibreCAD #1036: close function for polylines - keyboard short cut / button / explanation

WolfgangFahl, 2018-11-06:
### Expected behavior
When drawing a polyline with the mouse it should be obvious / intuitive how to close the polyline.
E.g. a right mouse click could to this.

### Observed behavior
ESC works to finish a polyline with not closing.
There is no close button no keyboard shortcut explained in the wiki

### Steps to reproduce or sample file
https://wiki.librecad.org/index.php?title=Starting_to_draw
### Operating System and LibreCAD version info

WolfgangFahl, 2018-11-06:
see http://forum.librecad.org/how-to-create-closed-polygons-td5707144.html that this has been asked before. I couldn't find the anwer.

lordofbikes, 2018-11-07:
The close polyline button is in the tool option bar which is active while drawing a polyline.
[image]

The tool option bar is dynamically and vary with the used tool, but not all tools have options.

Right click has ESC functionality while drawing for all tools and cancel the active tool.

closed 2018-11-07 by lordofbikes

WolfgangFahl, 2018-11-07:
What is the tool option bar and where is it under Mac OS X? It seems to also not be visible by default in the linux version - my ubuntu 16.04 LTS installed version 2.0.9.

lordofbikes, 2018-11-08:
Possibly the tool option bar was disabled by accident. Activate it here:
[image]

Then the tool options should be shown there:
[image]