LibreCAD/LibreCAD #1038: CVE-2018-19105 - Write access violation resulting in DOS

spotrh, 2018-11-08:
LibreCAD allows attackers to cause a denial of service (0x89C04589 write access
violation and application crash) or possibly have unspecified other impact via
a crafted file.

References:

https://code610.blogspot.com/2018/11/crashing-librecad-213.html

=====

I can confirm that the .dxf file referenced there crashes 2.2.0 rc1 as well.

turtle0x1, 2018-11-09:
Has this been reported properly (and fixed ?) or are you just dumping an exploit because "fun"

spotrh, 2018-11-09:
To be clear, I am not the author for this blog post, nor involved in any way with the exploit. I was informed by Red Hat's Infosec team that a CVE had been opened against Librecad (a package I maintain in Fedora) and investigated. When I discovered that there was no open issue against this potential vulnerability (already public), I filed one for tracking.

lordofbikes, 2018-11-11:
@spotrh, this fix is for our local copy of libdxfrw. 
The originator of libdxfrw is not responsive and the sf.net repo not maintained yet.
Can you use this fix for the Fedora package or can I support you in any way?
Our plans are to continue libdxfrw under LibreCAD, to guarantee future support.

Any thanks for reporting!

lordofbikes, 2018-11-12:
reopened because of #1045

lordofbikes, 2018-11-15:
Closing this, as functionality is confirmed in #1045.

commit 6da7cc5f (2018-11-11): fixed #1038, write access violation when a malicious dxf is fed to libdxfrw
closed 2018-11-15 by lordofbikes