LibreCAD/LibreCAD #1039: Negative Coordinate Issue

labusker, 2018-11-08:
### Expected behavior
I just loaded the latest software and would expect all menu & line commands to work.

### Observed behavior
Cannot use the command line to draw in a negative direction. One person on Github did mention that
using math to produce a negative direction works. Example: line; 0,0; 10,10; 20,20; 30,20-30.

### Steps to reproduce or sample file
Error example attached.
[image]

labusker, 2018-11-08:
On the same subject, if I wanted to rotate an object 20 degs clockwise, I could not just enter "-20", I would have to enter "20-40" which equals -20.

lordofbikes, 2018-11-08:
This is a duplicate of #981, please find a workaround there.
Also the forum discussion is linked there.
This one is hard to battle, because it is not reproducible yet.

closed 2018-11-08 by lordofbikes