LibreCAD/LibreCAD #1040: Move to Bottom Command

labusker, 2018-11-08:
### Expected behavior
Moving objects to top or bottom should work
### Observed behavior
I can move an object that is behind another to the top, I cannot move an object that is on top to the bottom. I also cannot raise over entity or lower after entity. What am I doing wrong?
### Steps to reproduce or sample file
To try to move an object to the bottom, I 1st select the top object, then  I press the "move to bottom" icon. Nothing will happen when I do this.
### Operating System and LibreCAD version info
[image]

dedicate617, 2018-12-10:
1. select the top object;
2. "move to bottom"
3. press "enter"
4. press "esc"
then it happend
--good luck

lordofbikes, 2020-06-01:
I'll reopen this, because the #1200 fix still leave room for improvement.
The issue before was, that it was not clear what to do to achieve the reorder.
It was simply a missing ENTER to accept the already selected entities.
Now it works without ENTER, but after the command it is not clear that something has happened, because the selection is still active.
I think after the reorder action has finished, the selected entities should be unselected to make the change immediately visible,

leo-smi, 2021-03-05:
The shortcut works, but when we copy the element to another place the element comes back to the main layer position. I mean, the copied element layer position is defined by the main layer, and do not keep the original element layer position.

PR #1200 (2020-06-01): Fix for 'move to bottom' command with existing selection
Fixes #1040. The "Move to Bottom" command now works when there is an existing selection. Demo below:

[image]
commit e6e1d429 (2020-06-01): Fix for 'move to bottom' command with existing selection.

Fixes #1040.