LibreCAD/LibreCAD #1047: Explode button state in Modify tool bar is inconsistent

lordofbikes, 2018-11-16:
### Expected behavior
After an explode action, the related button in Modify tool bar should be released.

### Observed behavior
The button stays in checked state, but the command line shows `Command:`, which means, that the explode action has finished.
Also the mouse cursor is reset from selection cross to arrow pointer.

### Steps to reproduce or sample file
Draw a rectangle or text or insert a block
Select the Explode tool from Tools -> Modify menu
Select the previous drawn entity and hit ENTER
The entity is exploded, the command line expects the next command and the mouse pointer is reset.
But the tool button stays checked.

### Operating System and LibreCAD version info
all OS
LibreCAD 2.2.0-rc1

lordofbikes, 2018-11-16:
This applies to the Modify menu too. Presumably, the action has to be unchecked, then the buttons from tool bar and menu are released.

lordofbikes, 2018-11-16:
This is a more common issue, many other single action tool in various tool bars are affected, e.g. Mirror, Revert direction or Create polyline from existing segments.

sand1024, 2026-09-26:
fixed in 2.2.2 (and latest 2.2.1)

closed 2026-09-26 by sand1024