LibreCAD/LibreCAD #1053: missing symbols needed for countersink, counterbore, and arc length

TheToric, 2018-12-02:
I would like to use unicode symbols sutch as [2335]⌵,[2322]⌢,[21a7]↧, and [2294]⊔ for dimentioning. However, the unicode font does not support all of these, and the unicode font does not support any of these, and the wqy-unicode fond only supports ↧, and does not support ⌀. All of these symbols are very commonly used in the mechanical drafting world.

ArchLinux, all updates performed
LibreCAD Version: 2.1.3
Compiler: GNU GCC 8.2.1
Compiled on: Nov  8 2018
Qt Version: 5.11.2
Boost Version: 1.68.0

shawncurry, 2019-05-25:
I don't believe this is a limitation of the LFF font format; it's likely the font simply doesn't have a representation for these characters.  An enterprising tinkerer might take a look at the unicode.lff in a text editor and add the entries they wanted.  The representation is basically like drawing a polyline.

Here's a little more info on the format:  [LibreCAD GitHub wiki](https://github.com/LibreCAD/LibreCAD/wiki/LFF-Font-Format).  You don't need to worry about the "bulge" and "sagitta" stuff unless you need arcs.

cryptoluks, 2020-05-31:
The Arch Linux package `ttf-dejavu` should provide fonts with these symbols you need.

app4soft, 2024-08-31:
I will take a look on adding this symbols into `unicode.lff`:

- `[2335] ⌵`
- `[2322] ⌢`
- `[21a7] ↧`
- `[2294] ⊔`

u2fly, 2025-04-16:
> As for Arc Lengt there are two possible variants of symbol representation
>- https://www.qcad.org/doc/qcad/latest/reference/en/scripts/Pro/Draw/Dimension/DimArcLength/doc/DimArcLength_en.html

For inline prefix dimension label needs to add `[2312] ⌒` (Arc) symbol, for ontop symbol over label needs to add `[2322] ⌢` (Frown) symbol — but LibreCAD has no option for setting ontop symbols over dimension label.

@dxli @sand1024

Also, related from Forum:
- https://forum.librecad.org/GD-amp-T-Geometric-Dimensioning-and-Tolerancing-td5723038.html
> Question:  Does Librecad have a utility to insert GD&T symbols - true position, datums, flatness etc. ? 

To simplify insert GD&T/Surface Finishing symbols in dimension label/text need to add listed bellow symbols with descriptions in _"Dimensions"_ dialog&toolbar's "Insert" list (and rename it to "Insert Symbol" list), and in _"Text"_/_"MText"_ dialogs' "Insert Symbol" list (as in "Insert Unicode" list symbols has no descrptions):

```
⌒ (Arc Length, inline)
⌢ (Arc Length, ontop)
⌓ (Segment)
⌔ (Sector)
❌ (Cross Mark / Datum Target Point)
⏤ (GD&T: Straightness)
⏥ (GD&T: Flatness)
○ (GD&T: Circularity / Roundness)
⌭ (GD&T: Cylindricity)
⌒ (GD&T: Line Profile)
⌓ (GD&T: Surface Profile)
⟂ (GD&T: Perpendicularity)
∠ (GD&T: Angularity)
∥ (GD&T: Parallelism)
⌖ (GD&T: Position)
⌯ (GD&T: Symmetry)
◎ (GD&T: Concentricity / Coaxiality)
↗ (GD&T: Circular Run-Out)
⌰ (GD&T: Total Run-Out)
Ⓐ (GD&T: Derived Feature)
Ⓒ (GD&T: Minimax Feature)
Ⓔ (GD&T: Envelope Requirement)
Ⓕ (GD&T: Free State)
Ⓖ (GD&T: Least Squares Feature)
Ⓘ (GD&T: Independency)
Ⓛ (GD&T: Least Material Condition)
Ⓜ (GD&T: Maximum Material Condition)
Ⓝ (GD&T: Minimum Circumscribed Feature)
Ⓟ (GD&T: Projected Tolerance Zone)
Ⓢ (GD&T: Regardless of Feature Size)
Ⓣ (GD&T: Tangent Plane)
Ⓤ (GD&T: Unequal Disposed Profile)
Ⓧ (GD&T: Maximum Inscribed Feature)
▷ (GD&T: Translation)
△ (GD&T: Dynamic Profile Tolerance Zone)
↔ (GD&T: Between)
→ (GD&T: From-To)
℄ (GD&T: Center Line)
⌱ (GD&T: Dimension Origin)
□ (GD&T: Square)
⌳ (GD&T: Slope)
⌲ (GD&T: Conical Taper)
⌴ (GD&T: Counterbore / Spotface)
⌵ (GD&T: Countersink)
↧ (GD&T: Depth / Deep)
```

>[!NOTE]
>Not all `ASME Y14.5`/`ISO 1101` symbols has Unicode symbols, some GD&T symbols are few latin letters/symbols combination (ligature), but here are special one of such:
>- `n/a (GD&T: All Over)` - like `⌮ (GD&T: All Around-Profile)` but diagonally-inverted with double-circle; both symbols used only as "Datum Feature" leader mid-point in a form of `⚬` and `⌾` label accordingly
>- `n/a (GD&T: Dimension Not To Scale)` - underlined dimension label?
>- `CF (GD&T: Continuous Feature / Contacting Feature)` - letters inside widened hexagon
>- `ST (GD&T: Statistical Tolerance)` - letters inside widined hexagon

u2fly, 2025-04-16:
> I don't believe this is a limitation of the LFF font format; it's likely the font simply doesn't have a representation for these characters.

These symbols are in `unicode.lff` now, and LibreCAD shows `⌒⌴⌵↧` correctly in MText/Text entities.

But in Dimension label entities `⌒⌴⌵↧` symbols LibreCAD shows them as `����` ("rombus").