LibreCAD/LibreCAD #1054: cluttered interface elements on HiDPI screens

AlexeyGusev, 2018-12-02:
### Expected behavior
Interface is expected to be aware of HiDPI-related scaling and interface elements should be displayed identically to non-HIDPI setup.
### Observed behavior
This is how the properties window looks:
[image]
To see all elements properly, you need to manually drag the window corner to increase window size. Very annoying when done frequently. Once manually enlarged, properties window looks like this:
[image]
Images below show broken interface elements displayed on properties toolbar, notice how the «Length», «Snap Point» and «Radius» are rendered:
[image]
[image]
Degree of distortion depends on scaling factor, scaling up <1.2 times introduces just marginally cluttered interface. Scale up 200% and the elements described above are not usable at all.
### Steps to reproduce or sample file
Scale up your display via gnome settings (Devices → Monitors → Scaling), KDE (System Settings → Display and Monitor → Scale Display) or xrandr --scale 0.Nx0.N, where N = {5…9}. Draw a line or circle, choose Tools → Modify → Properties. Observe properties window. Start drawing a line or a circle, observe properties toolbar above the drawing area.
### Operating System and LibreCAD version info
Version: 2.1.2
Compiler: GNU GCC 6.2.0
Compiled on: Sep 17 2016
Qt Version: 5.9.5
Boost Version: 1.62.0
System: Ubuntu 18.04.1 LTS

Gnome shell scaled to 200%, then xrandr --scale 1.25x1.25 (scale up twice, then scale down 1.25X)

Zooce, 2020-08-03:
I'm not an expert in Qt applications, but launching LibreCAD from the terminal like this seems to work:
```
$ QT_AUTO_SCREEN_SCALE_FACTOR=1 librecad &
```
I got this from: https://doc.qt.io/qt-5/highdpi.html.

archerdm, 2021-07-13:
I can confirm the UI issues noted by the OP. I'm using a 4K monitor set for 172 dpi without any screen scale factors. With the exception of text getting clipped short in many places as shown in the OP, the UI is otherwise well proportioned. QT_AUTO_SCREEN_SCALE_FACTOR does not solve the issue because it scales the graphics window as well, making the drawing pixelated and defeating the purpose of a 4K screen.

fa201, 2021-07-15:
@archerdm Which OS do you work on ?

archerdm, 2021-07-15:
I should have mentioned. Artix Linux, KDE 5.22.3, Qt 5.15.2, LibreCAD compiled from latest master branch.

rmsc, 2023-09-29:
> I'm not an expert in Qt applications, but launching LibreCAD from the terminal like this seems to work:
> 
> ```
> $ QT_AUTO_SCREEN_SCALE_FACTOR=1 librecad &
> ```

This does fix the toolbars in a way, but:

- The icons get all blurry unless their size is set to 32 in Widget Options
- The workspace gets pixelated

[image]

dongxuli-concord-design, 2023-09-29:
I feel this is getting high priorities.

I don't have any 4k display available for testing. I will borrow a macpro to test. Hope the virtual machine on it also allows windows testing