LibreCAD/LibreCAD #1055: Switch to LibreDWG

XVilka, 2018-12-06:
Yesterday new version of LibreDWG library was released with a few good features
https://github.com/LibreDWG/libredwg/releases/tag/0.7

It would be very beneficial to switch to this library finally.

new API:
  * add dwg_get_OBJECT(dwg) API for all objects in a DWG.
    analog to dwg_get_ENTITY, but including all objects, tables,
    block headers, even if unowned.
  Note that the dwg_api will be revamped from static to dynamic before 1.0.
  field accessors will be by name argument and ... (va_args)
  See the work/dynapi branch #59 (in progress).
  dwg_get_OBJECT(dwg) will be renamed to dwg_getall_OBJECT(dwg),
  dwg_get_ENTITY(blkhdr) to dwg_getall_ENTITY(blkhdr).

API breaking changes:
  * Changed BITCODE_RC from signed to unsigned char (#44)
  * LEADER changes: rename hooklineonxdir to hookline_dir,
    rename unknown_bit_4 to hookline_on,
    endptproj only until r2007. (but still not 100% correct)
  * MULTILEADER changes: rename num_vertex to numpts (analog to LEADER),
    change vertex to points[] (PR #49 Denis Pryt)
  * Fixed dwg_get_ENTITY(hdr): NULL-terminate the result (#60)

Important bugfixes:
  * Fixed all memory leaks and double-free's (#43, #42, PR #58 Denis Pryt)
  * Fixed EED sizes (PR #48, Denis Pryt)
  * Fixed BLOCK_HEADER.num_inserts, esp. with xref's (#44).
    For decode and encode.
  * Fixed LTYPE.dash (#44)
  * Fixup LEADER DXF values and path_type with annot_type mixup.
  * Fixed TABLE.value.data_type with flags kGeneral (#54 Denis Pryt)
  * Fixes VECTOR_CHKCOUNT segfault on unassigned obj->dxfname's
    (esp. for <=R12), and fix overflows (#56)
  * dwglayers: fix for DICTIONARY breaking the list output (#51)
  * dxf: fix dxf_blocks_write for empty paper_space (#52)
  * Fixed example/load_dwg.py allowing non-critical error codes

Other newsworthy changes:
  * Check for RUNNING_ON_VALGRIND and call dwg_free then, even
    if previously skipped.
  * Renamed all remaining API "null malloc" error messages to
    "Out of memory"
* Harmonized in_dxfb with in_dxf, but still WIP

rnbdc, 2018-12-06:
Considering the current DXF output from LibreCAD is unreadable in AutoCAD I would say this could be a nice change.
At least LibreCAD files can be read by the Open Design Alliance libraries... so you can open them in other programs that use such libraries (many commercial CADs) and save them again, anf then they will open on AutoCAD, but it would be nice if LivreCAD output was ok to start with.

Mannshoch, 2019-01-03:
Is that maybe better placed in https://github.com/LibreCAD/LibreCAD_3 ?

XVilka, 2019-01-03:
@Mannshoch done https://github.com/LibreCAD/LibreCAD_3/issues/207

lordofbikes, 2019-01-03:
For LibreCAD 2 I can close this, as the license incompatibility still exists. This was detailed discussed years ago on various places.
LibreDWG license hasn't changed and LibreCAD 2 is still bound to QCAD license.

closed 2019-01-03 by lordofbikes

Mannshoch, 2019-01-03:
Thanks for this important Information.
Could you may add this Licence Issue Information to the Readme and may Link this discussion and this Issue as example?

lordofbikes, 2019-01-03:
LibreCAD 2 is forked from old QCAD code, which is licensed GPLv2. The LibreCAD 2 code base has no relation with QCAD GPLv3 code.
The incompatibility is between QCAD GPLv2 and LibreDWG GPLv3+.
There were efforts and communication with both, QCAD and LibreDWG maintainers, but both rejected or ignored our requests to re-license their code.

Former issues are #351 and #392.

Some direct quotes from 2012 could be found here:
http://libregraphicsworld.org/blog/entry/whats-up-with-dwg-adoption-in-free-software
http://libregraphicsworld.org/blog/entry/libredwg-controversy-2-months-later
http://libregraphicsworld.org/blog/entry/libredwg-drama-the-end-or-the-new-beginning

All that doesn't apply to LibreCAD 3, because it is build from scratch and has no relation to any QCAD code.

liufeijin, 2020-06-08:
Hope LibreCAD2 and LibreCAD3 can change to LibreDWG,
now it can't be used at all on  importing DWG

u2fly, 2025-04-22:
> LibreDWG license hasn't changed and LibreCAD 2 is still bound to QCAD license.

@lordofbikes @dxli

Is not _"or (at your option) any later version"_ allows you switch LibreCAD 2 to GPLv3 as to later version of GPL?
https://github.com/LibreCAD/LibreCAD/blob/6a138269b15f0cb4f6b2b0459a4888343ed122b5/licenses/gpl-2.0.txt#L296-L299

As there was no "GPLv2 (only)" stated in QCAD 2.0.5 CE it allows switch LibreCAD to later version of GPL
- http://www.ribbonsoft.com/archives/qcad/qcad-2.0.5.0-1-community.src.tar.gz

Also, from forum
- https://forum.librecad.org/Read-AutoCAD-s-DWG-file-format-tp5724062p5726971.html
> There should already be such an exception there for Qt, since it is dual licensed under QT commercial license and GPLv3+ afaik, neither of which are compatible with GPLv2