LibreCAD/LibreCAD #1062: join libreoffice community

techsupp82, 2018-12-23:
I'm going straight to the point, we need a good alternative to autocad, the project has progressed very slowly lately, allows it to be part of the LibreOffice community, with all the advantages, also imitate the UI ribbon of Autocad 2019

### Expected behavior

### Observed behavior

### Steps to reproduce or sample file

### Operating System and LibreCAD version info

MikePryadko, 2020-02-04:
@techsupp82 UI ribbon sucks! :smiling_imp: 
AutoCAD is not very bad software, but not very good: better look at SolidWorks or [Компас-3D](https://kompas.ru/) (Russian software, better than AutoCAD, but a little bit worse than SolidWorks) :relieved:

natrius, 2021-04-23:
Well, luckily the UI ribbon is optional after all. Would sound like a good step and maybe would bring more attention and maybe developers to help?

fitojb, 2023-01-09:
I don’t think, as a member of The Document Foundation, that it would be a good fit for this project (given its current governance model), but we can be friends and collaborate :)

techsupp82, 2023-01-09:
hahaha

natrius, 2023-01-09:
> governance model

Out of curiosity - what do you mean with that?

techsupp82, 2023-01-09:
por curiosidad  = out of curiosity (usamos la misma expresión) (we use the same expresion)

many times many free software teams drown in their own ideologies, when the main goal should always have been to gain more users, if there are more users there will be more resources... like blender for example. I've been a friend of free software for decades, I'm just stating the obvious.

I'm from Cordoba, we make jokes, thanks for the sense of humor

fa201, 2023-08-09:
5 years, we did not join LibreOffice community.
Should we close this and provide a statement to explain ?

techsupp82, 2023-08-09:
No,  As the project grows and gains more adoption, there is not just one path. Best of luck.
Thanks!