LibreCAD/LibreCAD #1064: LibreCAD 2.2.0.rc1, Compilation stops prematurely

JohnCC330, 2019-01-04:
### Expected behavior
Complete compilation
### Observed behavior
Compilation stops prematurely
### Steps to reproduce or sample file
(see below)
### Operating System and LibreCAD version info
LibreCAD 2.2.0.rc1
Qt-5.11.3 
Slackware64-current

Compilation was - with minor warning - Ok till:

```
g++ -c -pipe -g -fext-numeric-literals -O2 -O2 -fPIC -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DDWGSUPPORT -DQC_APPDIR="librecad" -DLC_VERSION="2.2.0-alpha" -DQT_NO_DEBUG -DQT_SVG_LIB -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I. -I/usr/include -I../../libraries/libdxfrw/src -I../../libraries/jwwlib/src -Icmd -Ilib/actions -Ilib/creation -Ilib/debug -Ilib/engine -Ilib/fileio -Ilib/filters -Ilib/generators -Ilib/gui -Ilib/information -Ilib/math -Ilib/modification -Ilib/printing -Iactions -Imain -Itest -Iplugins -Iui -Iui/forms -Iui/generic -I../res -I/usr/include/qt5 -I/usr/include/qt5/QtSvg -I/usr/include/qt5/QtPrintSupport -I/usr/include/qt5/QtWidgets -I/usr/include/qt5/QtGui -I/usr/include/qt5/QtCore -I../../generated/librecad/moc -I/usr/include/libdrm -I../../generated/librecad/ui -I/usr/lib64/qt5/mkspecs/linux-g++ -o ../../generated/librecad/obj/qg_coordinatewidget.o ui/forms/qg_coordinatewidget.cpp
ui/forms/qg_commandwidget.cpp: In constructor ‘QG_CommandWidget::QG_CommandWidget(QWidget*, const char*, Qt::WindowFlags)’:
ui/forms/qg_commandwidget.cpp:59:60: error: invalid use of incomplete type ‘class QAction’
     auto a1 = new QAction(QObject::tr("Keycode Mode"), this);
                                                            ^
In file included from /usr/include/qt5/QtWidgets/QApplication:1:0,
                 from ../../generated/librecad/ui/ui_qg_commandwidget.h:14,
                 from ui/forms/qg_commandwidget.h:29,
                 from ui/forms/qg_commandwidget.cpp:26:
/usr/include/qt5/QtWidgets/qapplication.h:225:18: note: forward declaration of ‘class QAction’
     friend class QAction;
                  ^~~~~~~
ui/forms/qg_commandwidget.cpp:62:27: error: incomplete type ‘QAction’ used in nested name specifier
     connect(a1, &QAction::toggled, this, &QG_CommandWidget::setKeycodeMode);
                           ^~~~~~~
ui/forms/qg_commandwidget.cpp:72:65: error: invalid use of incomplete type ‘class QAction’
     auto a2 = new QAction(QObject::tr("Load Command File"), this);
                                                                 ^
In file included from /usr/include/qt5/QtWidgets/QApplication:1:0,
                 from ../../generated/librecad/ui/ui_qg_commandwidget.h:14,
                 from ui/forms/qg_commandwidget.h:29,
                 from ui/forms/qg_commandwidget.cpp:26:
/usr/include/qt5/QtWidgets/qapplication.h:225:18: note: forward declaration of ‘class QAction’
     friend class QAction;
                  ^~~~~~~
ui/forms/qg_commandwidget.cpp:73:27: error: incomplete type ‘QAction’ used in nested name specifier
     connect(a2, &QAction::triggered, this, &QG_CommandWidget::chooseCommandFile);
                           ^~~~~~~~~
ui/forms/qg_commandwidget.cpp:76:71: error: invalid use of incomplete type ‘class QAction’
     auto a3 = new QAction(QObject::tr("Paste Multiple Commands"), this);
                                                                       ^
In file included from /usr/include/qt5/QtWidgets/QApplication:1:0,
                 from ../../generated/librecad/ui/ui_qg_commandwidget.h:14,
                 from ui/forms/qg_commandwidget.h:29,
                 from ui/forms/qg_commandwidget.cpp:26:
/usr/include/qt5/QtWidgets/qapplication.h:225:18: note: forward declaration of ‘class QAction’
     friend class QAction;
                  ^~~~~~~
ui/forms/qg_commandwidget.cpp:77:27: error: incomplete type ‘QAction’ used in nested name specifier
     connect(a3, &QAction::triggered, leCommand, &QG_CommandEdit::modifiedPaste);
                           ^~~~~~~~~
ui/forms/qg_commandwidget.cpp: In destructor ‘virtual QG_CommandWidget::~QG_CommandWidget()’:
ui/forms/qg_commandwidget.cpp:90:52: error: invalid use of incomplete type ‘class QAction’
     settings.setValue("Widgets/KeycodeMode", action->isChecked());
                                                    ^~
In file included from /usr/include/qt5/QtWidgets/QApplication:1:0,
                 from ../../generated/librecad/ui/ui_qg_commandwidget.h:14,
                 from ui/forms/qg_commandwidget.h:29,
                 from ui/forms/qg_commandwidget.cpp:26:
/usr/include/qt5/QtWidgets/qapplication.h:225:18: note: forward declaration of ‘class QAction’
     friend class QAction;
                  ^~~~~~~
ui/forms/qg_commandwidget.cpp:90:54: error: invalid use of incomplete type ‘class QAction’
     settings.setValue("Widgets/KeycodeMode", action->isChecked());
                                                      ^~~~~~~~~
In file included from /usr/include/qt5/QtWidgets/QApplication:1:0,
                 from ../../generated/librecad/ui/ui_qg_commandwidget.h:14,
                 from ui/forms/qg_commandwidget.h:29,
                 from ui/forms/qg_commandwidget.cpp:26:
/usr/include/qt5/QtWidgets/qapplication.h:225:18: note: forward declaration of ‘class QAction’
     friend class QAction;
                  ^~~~~~~
In file included from /usr/include/qt5/QtCore/qvariant.h:51:0,
                 from /usr/include/qt5/QtCore/QVariant:1,
                 from ../../generated/librecad/ui/ui_qg_commandwidget.h:12,
                 from ui/forms/qg_commandwidget.h:29,
                 from ui/forms/qg_commandwidget.cpp:26:
/usr/include/qt5/QtCore/qobject.h: In instantiation of ‘T QObject::findChild(const QString&, Qt::FindChildOptions) const [with T = QAction*; Qt::FindChildOptions = QFlags<Qt::FindChildOption>]’:
ui/forms/qg_commandwidget.cpp:89:55:   required from here
/usr/include/qt5/QtCore/qobject.h:165:51: error: incomplete type ‘ObjType {aka QAction}’ used in nested name specifier
         return static_cast<T>(qt_qFindChild_helper(this, aName, ObjType::staticMetaObject, options));
                               ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [../../generated/librecad/obj/qg_commandwidget.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory `/usr/local/src/graphics/LibreCAD-2.2.0-rc1/librecad/src'
make[1]: *** [sub-src-make_first] Error 2
make[1]: Leaving directory `/usr/local/src/graphics/LibreCAD-2.2.0-rc1/librecad'
make: *** [sub-librecad-make_first-ordered] Error 2

```

lordofbikes, 2019-01-05:
Build succeeds with these modifications:
http://forum.librecad.org/LibreCad-2-2-0-rc1-Mods-to-compile-with-Qt-5-11-0-td5716030.html

JohnCC330, 2019-01-05:
Yes it does - great and thanks!

lordofbikes, 2019-01-05:
I'll keep it open, we have to apply this modifications to the code base.

shawncurry, 2019-05-25:
commit 6c392e90

closed 2020-03-07 by lordofbikes