LibreCAD/LibreCAD #1066: Neither Tools->Modify nor Icons allow selecting anything

ModifyProblem, 2019-01-05:
### Expected behavior - When I try to modify things a hatched cross hair appears but will not let me select anything to modify.  This occurs whether using the text choice options or the icons in the tool bar.

### Observed behavior - Nothing is selected

### Steps to reproduce or sample file - ?

### Operating System and LibreCAD version info Librecad Version: 2.1.3
Compiler: GNU GCC 4.9.1
Compiled on: Sep 23 2016
Qt Version: 5.4.1
Boost Version: 1.60.0
System: Windows 
Computer - Windows 10 Home Edition

lordofbikes, 2019-01-05:
No valuable information to understand this issue.
Sounds like a usage problem, maybe the [forum](http://forum.librecad.org) is a better place to ask for support.

closed 2019-01-05 by lordofbikes

ModifyProblem, 2019-01-06:
"No valuable information to understand this issue."
OK, apparently your definition of valuable and mine differ.  And your inability to articulate what kind of information you consider "valuable" does nothing towards understanding this problem.
Apparently this forum is about as useful as the government.

ModifyProblem, 2019-01-06:
P.S. It is this forums submission form I filled out.  So, like the government you demand something and then say well - that's not what I wanted.  Politics the whole way down.