LibreCAD/LibreCAD #1070: Copy and insert between files

mitscherdinger, 2019-01-26:
Dear Developers!
I have experienced problems copying parts of a FreeCAD generated DXF into a new LibreCAD file. Don't know if this is a FreeCAD or a LibreCAD issue, but I thought I should let you know…

### Observed behavior
Ctrl+A, Ctrl+C, Select Reference, switch to an empty LibreCAD file and insert doesn't copy all stuff, only one element is inserted. Meanly, LibreCAD first shows the expected behaviour, but switching to another window (inside LibreCAD) or just saving, closing and re-opening shows that the job isn't done.

### Steps to reproduce or sample file
I dropped a file on my server to reproduce:
https://www.riotmusic.de/linux-audio/TraktrixFreeCADExport_I.dxf

### Operating System and LibreCAD version info
Version: 2.1.3
Compiler: GNU GCC 8.2.0
Erstellt am: Sep 17 2018
Qt Version: 5.11.3
Boost Version: 1.62.0
System: Debian GNU/Linux buster/sid

shawncurry, 2019-05-25:
Please explain a little further exactly what you're trying to do.  Are you trying to copy from FreeCAD's window, into LibreCAD's window?  I don't think this is a supported feature, as FreeCAD and LibreCAD would somehow need to agree to some standard clipboard protocol.

Otherwise, it appears that I can copy and paste in LibreCAD just fine.  The crosshair in the center of the original drawing of course wouldn't be copied; that's the indicator for the origin point of the drawing.

[image]
[image]

mitscherdinger, 2019-05-26:
> Are you trying to copy from FreeCAD's window, into LibreCAD's window?
No. I exported a dxf from freecad (f.e. the one I linked in my bug report), opened that in librecad and tried to copy and paste inside librecad. Or more precise, from one librecad window to another librecad window.
I made a screenshot of the result, but I couldn't upload it. Here's a link:
http://riotmusic.de/linux-audio/2019-05-26%2011-52-53.png

Left side is the original, right side is the attempt to copy from one window to another.

> Otherwise, it appears that I can copy and paste in LibreCAD just fine. The first pic is pasting into the same drawing, the 2nd is a new drawing. The crosshair in the center of the original drawing of course wouldn't be copied; that's the indicator for the origin point of the drawing.

Mmmh. Then it's the debian package that seems to have problems, here…
Here are the recent specs:
Version: 2.1.3
Compiler: GNU GCC 8.3.0
Erstellt am: May 16 2019
Qt Version: 5.11.3
Boost Version: 1.67.0
System: Debian GNU/Linux 10 (buster)