LibreCAD/LibreCAD #1074: LibreCAD will not open AutoCAD 2019 .dwg files

zuron, 2019-02-12:
LibreCAD will not open the newest AutoCAD 2019 dwg files.
 but will open the AutoCAD  2017 dwg or dxf files.

tenbob, 2022-02-26:
The main point about this bug is LibreCAD is displaying the wrong error message.

If it said "Can't open this file - incompatible DWG version", then most people would know what to do.  

At the moment it's sending people on a wild goose chase by telling them the file doesn't exist or has the wrong persmissions.

 Can the developers fix the error message for a quick win.

lordofbikes, 2022-02-28:
@tenbob misleading error message was caused by [libdxfrw](https://github.com/LibreCAD/libdxfrw), there was only `true`/`false` return code, no room for interpretation :rofl:.
This is fixed in latest code:
edcc7c856cf8541a6b560b4fe8f86a49307691cf
ec8b1a589e319f5535e34666dddb263c9f96a047

This issue with AutoCAD 2019 (AC1032) files still exists, but the error messages is now `unsupported DXF/DWG file version`