LibreCAD/LibreCAD #1075: If there are blank lines in dxf files, librecad will parse less blocks

CodingGou, 2019-02-14:
If there are blank lines in dxf files, librecad will parse less blocks,that is say some blocks won't be parsed.

lordofbikes, 2019-02-14:
So the question is, where do empty lines in DXF come from?
The DXF definition is clear, group codes and values in alternating lines.
There is no talk about empty lines.
Do you have a sample?

CodingGou, 2019-02-18:
So far, I have only encountered one, the original dxf file really should not have a blank line, but this time I did encounter it.

CodingGou, 2019-03-01:
but sometimes if we delete blank lines ，it will not parse any block. In this example
[118419F1901.dxf.zip](https://github.com/LibreCAD/LibreCAD/files/2918066/118419F1901.dxf.zip)

[image]

lordofbikes, 2019-03-01:
In the second file the blank line is correct.
It is in STYLE table, the '  4' in line 39 is the code for a string and in line 40 is an empty string.
When you delete line 40, the '  0' becomes the string for code '  4' and 'ENDTAB' from line 42 becomes the next group code. As 'ENDTAB' is invalid for group codes, the file becomes useless.

For the first file the situation is different, there are a lot of malformed strings like these:
[image]
Line 2431 is the group code '  1' expecting a string.
Line 2432 is a string, that's correct so far.
But in line 2433 should be the next group code.
This makes code/values pairs impaired and misinterprets following lines until the next blank line make them pairing again.

So basically blank lines are not an issue. I created a layer with empty name which results in an empty line too, but this loads correct. Also your second sample file load correct for me.

For the first file it is correct, that the blank lines are causing the issue. But beside that, the file has some more issues. There are some important sections missing, like HEADER or TABLES with line styles and layers. Means the creating program was causing this issue by creating not compatible DXF files.
In my estimation we should not try to fix this in our DXF reader as this may produce side effects for correct DXF files and that would be catastrophic.

It seems that this is a rare exception you have to handle manually.
If you have a bunch of these files I recommend to write a small console application for batch detection and correction of these files to make them DXF conform.
See https://www.autodesk.com/developer-network/platform-technologies/autocad-dxf-archive for various DXF version references. Your first file has no version information at all, the second one is AC1009 which is R11/R12 format. The archive starts with R13, but that should not matter. The main difference iirc is that R12 didn't support ellipse or spline entities, the format itself didn't change, only new group codes were added.
Don't hesitate to ask if anything is vague.

closed 2019-03-01 by lordofbikes