LibreCAD/LibreCAD #1082: Box joint modification tool

runningopenloop, 2019-03-21:
### Overview

Hello, I’m suggesting a box joint tool.  This would allow the user to do designs such that after cutting on CNC could then be assembled with the provided box joints.  My primary use case for CAD is to create parts on my Maslow CNC.  This tool would help facilitate that.

My thought was the user should be able to add box joints to existing drawings.  This way when making the part you can focus on the part and not all these joints.

### Crazy idea
I had originally thought about allowing the user to...
(1) define edges which should be seamed together,
(2) define info about how the box joints are created, then
(3) ‘execute’ the application of box joints

### More realistic idea
However I think this is more work that it’s worth and maybe not as useful as a simpler modification tool which can be used on existing lines.

When selecting a box joint modification (as proposed, open to re-direction), the menu consists of:
* finger size
* finger depth
* finger or slot first.
* Checkbox (or combobox) for auto add of dogbones.  Technically there would be a few possible types of dogbones which would apply here, hence consideration for a combobox.
	*  dogbone radius is another option here.
            * I’m nearly done added a dogbone tool, so this code could be re-used.

#### Adding box joint procedure
* The first click would select the line to operate on.
	* Also defines starting point for operation?
* The second click would indicate if the joint should be added in an additive fashion or subtractive fashion depending if the second click was on one side of the line or the other.

#### Extra behavior for first click is to also define starting point of line:
As for the first finger or slot.  The start point of the line as-is could be used, or the line start/end point could be selected as a starting point for the box joint depending on proximity.

#### Dealing with length % finger size != 0.
Given the size of the finger, there may need to be some user input or intelligence on how to handle lines which where length % finger size != 0.  With minimal thought, my first reaction would be to have the first finger or slot be bigger to address this.  This means the minimum size for the box joint tool to have any effect would be 2 * finger size as the first slot finger could be as large as < 2*finger size.

#### Other thoughts
* Should this be a modification, or a plug-in, or ?
* In the same fashion that the exiting fillet draws the to be added fillet before commit, this tool should do the same.

### Closing
Thanks for your consideration and thoughts.

73,
Timothy

shawncurry, 2019-05-24:
This sounds like it could be a great idea for a shape generating plugin.  Perhaps the user could define the dimensions of the box, number of sides, etc, and the tool would produce a complete pattern.  This could even be built upon to produce shapes for a specific purpose, such as HVAC fittings.

runningopenloop, 2019-05-25:
Hello Shawn,

Thanks for your thoughts, however I'm not thinking of this as a generation tool but a modification tool.  It modifies existing drawn items to be able to use finger joints to connect.  I've gotten stalled on finishing the dogbone modification.  But hope to resume that soon, then take this up.

Note bCNC has method to generate a box with finger joints.

73,
Timothy