LibreCAD/LibreCAD #1083: Vote for a new release.

Mannshoch, 2019-03-21:
Please focus for a new release.
Thanks

veronyka, 2019-03-31:
Please work on new release, Draftsight is no longer free, you guys are the only one now.

slokhorst, 2019-04-06:
Seems like there are no more open issues in the [2.2.0 Milestone](https://github.com/LibreCAD/LibreCAD/milestone/3). Is there anything else holding back the release?

yves78, 2019-04-07:
Librecad is very useful for me : I use it often  for drawing parts before machining them, manually or with NC machine,  and I would really appreciate if the product could continue to be developped and maintained.

jd666, 2019-11-24:
It is now 19 month since the 2.2.0 RC.. will there ever be a new release?

AKmatiAK, 2019-12-28:
pls keep this great program alive :)

lordofbikes, 2019-12-29:
It is not a matter of voting for a new release, so I'll close this now.
It's a matter of circumstances that 2.2.0 wasn't released in 2019.
We gonna combine forces with the new year to finally release the new version.

closed 2019-12-29 by lordofbikes