LibreCAD/LibreCAD #1085: Zoom panning can only be used once and must then be reactivated again

michael0710, 2019-04-09:
While using zoom panning by clicking on the button in the toolbar...

### Expected behavior
...I want to pan as long as the zoom panning button is activated (Just like the button called Window Zoom)

### Observed behavior
...I can only pan once and as soon as I release the mouse button the button is still activated but I am not able to pan with a further click on the drawing area. I need to click on the (still activated) zoom panning button and then it will work again for one pan.

### Steps to reproduce or sample file
Just draw a simple line to be able to see something moving and it will work with every .dxf file (I am not sure if it works with other filetypes but I believe it won't.

### Operating System and LibreCAD version info
Version: 2.2.0-rc1
Compiler: GNU GCC 4.9.1
Compiled on: Feb 16 2018
Qt Version: 5.4.1
Boost Version: 1.53.0
System: Windows 10

shawncurry, 2019-05-24:
This appears to have been caused by some refactoring by dxli from 4 September 2011:  commit cc6bc9e2 "bug # 3348447, need to test more".  Handling of the middle-mouse button was removed from the mouseReleaseEvent handler.  However, this was a disambiguation necessary for proper handling of both middle-click panning, and the "Zoom Panning" (toolbar) mode.

For the case of middle-click panning, the mode should end when the mouse button is released.  For the toolbar mode, panning should return to the "open hand" state, until the mode is ended by pressing the ESC key or selecting Window Zoom mode.

pseudo-code:
```
mouseReleaseEvent(MouseButton b) {
  if (b == MouseButton::middle)
    endPanning();
  else
    resumePanning();
}
```
I have added this fix to my other pending commit.

lordofbikes, 2019-06-10:
Before the fix it was not needed to reset the Pan tool, as it only worked once.
Now it's necessary to reset the tool, when panning is finished.
With ESC reset works, but most tools in LibreCAD, if not all, could be reset by right click too.
Right click must be implemented to reset the Pan tool.

shawncurry, 2019-06-12:
Included in pull request #1125

PR #1122 (2019-06-10): Feature/issue1108
- **Fix #1108 Improve User Experience while working with multiple drawings**
- Fix #515 DWG zoom position on opening
- Fix #1085 Zoom panning can only be used once and must then be react
- Add File > Save All and File > Close All actions + icons
- This commit also includes my un-merged changes from #1099 (preferences stuff)
commit b5e21c59 (2019-06-10): - fix #1085 Zoom panning can only be used once and must then be reactivated again
- restored a disambiguation needed for proper handling of middle-mouse vs toolbar panning
commit e5aab73a (2020-03-08): fix #1085 Zoom panning can only be used once and must then be reactivated
- Right clicking with the mouse now exits the toolbar panning mode.
closed 2020-03-08 by lordofbikes