LibreCAD/LibreCAD #1086: improve autosave for unnamed files

lordofbikes, 2019-04-10:
### Expected behavior
On IRC channel an user raised the question, if new created files are auto saved too.
And when, where are they saved.

### Observed behavior
A quick look pointed the systems tmp folder where #Unnamed.dxf is auto saved.
It seems, that #Unnamed.dxf from previous session is overwritten.

### Steps to reproduce or sample file
- create a new drawing
- wait at least one auto save cycle
- check the tmp folder and the content of #Unnamed.dxf
- create a new drawing and repeat the steps above

### Conclusion
It would be nice to have some kind of ID to keep more than one #Unnamed.dxf.
And possibly a setting in application preferences for the location of #Unnamed.dxf files.
But we have to consider too, that this may fill the drive over time. So this possibly needs a mechanism to delete old #Unnamed.dxf after a while, or when the drawing is saved with a new name.
Any suggestions are welcome.

xanderdin, 2019-04-11:
What's the real idea behind that file? Is it only meant to keep some job after crash or something else?

There are obviously much more issues with that file.

On my system (Debian) it is created in /tmp directory. Let's look at the file's ownership and permissions:

```
ls -l /tmp/#Unnamed.dxf

-rw-r--r-- 1 user user 15847 Apr 11 18:57 /tmp/#Unnamed.dxf
```
It belongs to the current user, can be modified or deleted only by the current user, but can be read by anyone else. First of all, it is absolutely insecure.  Second, if I log out or just lock my session, and other user logs in on my computer and runs LibreCAD, some errors potentially may occur.

So, if we want to save that file in /tmp directory, it is a good idea to create a sub-directory, something like:
```
/tmp/LibreCAD-PID-USER
```
where PID is the process ID of the running LibreCAD instance, and USER is the username of the user running that instance. And that sub directory should also have only 700 access mode for the security reasons.

Why PID is also needed? Because a user can run two or more instances of LibreCAD simultaniously, and each instance can have several unnamed documents.

The documents themselves could be named according to their names in the LibreCAD, e.g.:
```
unnamed document 1  -->  #Unnamed-1.dxf
unnamed document 2  -->  #Unnamed-2.dxf
```
and so on.

Concerning disk overflow. On Debian, /tmp directory is mounted as tmpfs, which means that after system reboot it becomes empty. On other systems, I believe, the same or similar approach is used to prevent /tmp overflow.

**To summarize my point (for linux systems):**
```
/tmp/LibreCAD-PID-USER/#Unnamed-N.dxf
```
where
`PID` - process id of the running LibreCAD instance;
`USER` - username of the user running a LibreCAD instance;
`N` - serial number of _unnamed document_ in the current LibreCAD instance.

Set access mode for `/tmp/LibreCAD-PID-USER/` to be 700.

P.S.:  what about # sign, is it really needed?

fa201, 2019-04-11:
My understanding is the following : Autosave allows to retrieve the work in progress file in case of crash or unexpected event. So it protects important data, so this important file should have a name. If unname.dxf is not important then Autosave is not required. We could think of Autosave as launching a Save As dialog if there is no name : the user will name this file if it is important or cancel if it is not. If a name exists then nothing changes...